MatterHackers/MatterControl #4590: reset to defaults not clearing z-calibration and xy-calibration

larsbrubaker, 2019-05-30:
[image]

larsbrubaker, 2019-05-30:
This did reset correctly when I exported then imported the print with re-calibrate

[image]

larsbrubaker, 2019-05-30:
Not sure what I did to achieve this. Tried again and only got it to work.

jlewin, 2019-05-30:
Completed `Reset To Defaults` on a configured printer, all state reset as expected:
[image]

larsbrubaker, 2019-05-31:
I'll re-file if I can reproduce it.

closed 2019-05-31 by larsbrubaker