MatterHackers/MatterControl #4610: Purge not working correctly

mstange42, 2019-06-13:
Bug Report
============
**Steps to Reproduce**
Print anything

**Expected Behavior**
Purges correctly

**Actual Behavior**
When it does its original purge it stops at the end of the purge and pools a little then it moves around, usually back near the X,Y Origin, then goes to start the print. When it starts the skirt one of two things happen, first it is clogged. The second is that it doesn't start the skirt, then eventually goes after 2 to 3 inches of movement. When it does this it usually has strings that hold on and pull around. Then this stringing screws up the first brim edge or perimeter of the model.

**MatterControl Build Number**
2.19.6, current release version

**Operating System Version**

**Printer Make/Model**
Pulse D-211 w/ upgraded full metal hot end / Firmware for D-212 modified by Matterhackers

larsbrubaker, 2019-06-15:
Developer Notes:

We do not automatically suppress the M109 post start gcode if it is present in the start gcode.

larsbrubaker, 2019-06-15:
@mstange42  Thanks for the report of this bug. 
It has been fixed in source and will be in the next update.
We are planning to release an alpha update sometime around the end of the month.

closed 2019-06-21 by larsbrubaker

mstange42, 2019-06-21:
Thank you, Can't wait for a usable version, for me.