MatterHackers/MatterControl #4611: Empty bed will not print

AGmukbooks, 2019-06-13:
[CCR10_Octopus_v5.5_spiral_support_fixed_gcode.zip](https://github.com/MatterHackers/MatterControl/files/3288463/CCR10_Octopus_v5.5_spiral_support_fixed_gcode.zip)

all gcode files i have created from cura are giving me the error "empty bed: no printable part exists within the bounds of the printer bed. add content to continue
[image]

this has only begun after the latest update i have donwloaded yesterday

jlewin, 2019-06-14:
Thank you for reporting this issue. We will be making a hot fix for this ASAP and pushing it out.

larsbrubaker, 2019-06-14:
We have posted the fix to this, it is currently in the Alpha channel.
After testing, it will be promoted to release. This should take about  day.
The new build is: 2.19.6.10328

If you can test this and verify it is working for you that would be appreciated. You can find it in the Alpha channel.
Thank you!

[image]

[image]

AGmukbooks, 2019-06-14:
Thank you for your quick response! The error has been fixed and the "Start Print" button is now available.
[image]

closed 2019-06-26 by larsbrubaker