MatterHackers/MatterControl #4625: Unable to import items from Fusion360

meelo88, 2019-06-21:
Updated from 2.19.210169 to 2.19.6.10328 and I am no longer able to import my design in direct from Fusion360. I am still able to do so with Matterhackers 1.7.5.

Please advise.

Thanks!

jlewin, 2019-06-21:
What sort of errors are you running into?

The workflow for using meshes from Fusion360 should be the same as with any other 3D modeling package. To be sure I just went through the steps of exporting a body from Fusion360 to a new STL file and then opened and sliced the file in the latest version of MatterControl without issue:

[image]

meelo88, 2019-06-21:
In the previous version, I was able to use the make button to directly import my design into MatterControl and from there I could save it as an STL after modifying the angle or layer position.

That is what I have been using but I can confirm your method works with no issues.

[image]

jlewin, 2019-06-21:
It looks like we have a regression with file associations that's preventing this from working correctly. 

We'll get this fixed ASAP, thanks for the report.

In the meantime, you can use the `Save As STL` feature in Fusion to export the mesh to your file system:

[image]

then open in MatterControl:

[image]

meelo88, 2019-06-21:
Sounds good. Thanks!

closed 2019-06-26 by larsbrubaker