MatterHackers/MatterControl #4665: MacOS crash when exporting G-code

Marcus-Depew, 2019-09-18:
Bug Report
============
**Steps to Reproduce**
1. Bring in STL
2. Slice 
3. Export G-Code

**Expected Behavior**
"Save As" window pops up

**Actual Behavior**
MC crashes to desktop

**MatterControl Build Number**
2.19.7.10356

**Operating System Version**
macOS Mojave - 10.14.6

**Printer Make/Model**
Pulse S-232 / Pulse D-232

**Terminal Output / **

[ERROR] FATAL UNHANDLED EXCEPTION: System.NullReferenceException: Object reference not set to an instance of an object
  at MatterHackers.MatterControl.ExportPrintItemPage+<>c__DisplayClass3_0.<.ctor>b__0 (System.Object s, MatterHackers.Agg.UI.MouseEventArgs e) [0x0005b] in <ca1405d15e5d42b589cad20282092f68>:0 
  at MatterHackers.Agg.UI.GuiWidget.OnClick (MatterHackers.Agg.UI.MouseEventArgs mouseEvent) [0x0000a] in <f3a6caf3214c4942a581165cc935211a>:0 
  at MatterHackers.Agg.UI.GuiWidget.OnMouseUp (MatterHackers.Agg.UI.MouseEventArgs mouseEvent) [0x00256] in <f3a6caf3214c4942a581165cc935211a>:0 
  at MatterHackers.MatterControl.CustomWidgets.SimpleButton.OnMouseUp (MatterHackers.Agg.UI.MouseEventArgs mouseEvent) [0x00000] in <ca1405d15e5d42b589cad20282092f68>:0 
  at MatterHackers.Agg.UI.GuiWidget.OnMouseUp (MatterHackers.Agg.UI.MouseEventArgs mouseEvent) [0x00180] in <f3a6caf3214c4942a581165cc935211a>:0 
  at MatterHackers.Agg.UI.GuiWidget.OnMouseUp (MatterHackers.Agg.UI.MouseEventArgs mouseEvent) [0x00180] in <f3a6caf3214c4942a581165cc935211a>:0 
  at MatterHackers.Agg.UI.GuiWidget.OnMouseUp (MatterHackers.Agg.UI.MouseEventArgs mouseEvent) [0x00180] in <f3a6caf3214c4942a581165cc935211a>:0 
  at MatterHackers.Agg.UI.SystemWindow.OnMouseUp (MatterHackers.Agg.UI.MouseEventArgs mouseEvent) [0x00017] in <f3a6caf3214c4942a581165cc935211a>:0 
  at MatterHackers.Agg.UI.GuiWidget.OnMouseUp (MatterHackers.Agg.UI.MouseEventArgs mouseEvent) [0x00180] in <f3a6caf3214c4942a581165cc935211a>:0 
  at MatterHackers.Agg.UI.GuiWidget.OnMouseUp (MatterHackers.Agg.UI.MouseEventArgs mouseEvent) [0x00180] in <f3a6caf3214c4942a581165cc935211a>:0 
  at MatterHackers.Agg.UI.GuiWidget.OnMouseUp (MatterHackers.Agg.UI.MouseEventArgs mouseEvent) [0x00180] in <f3a6caf3214c4942a581165cc935211a>:0 
  at MatterHackers.Agg.UI.SystemWindow.OnMouseUp (MatterHackers.Agg.UI.MouseEventArgs mouseEvent) [0x00017] in <f3a6caf3214c4942a581165cc935211a>:0 
  at MatterControlMac.MacPlatformWindow.MouseUp (AppKit.NSEvent theEvent) [0x0004a] in <f6b6c4b89de3427ab6b4c2f9608ae461>:0 
  at (wrapper managed-to-native) AppKit.NSApplication.NSApplicationMain(int,string[])
  at AppKit.NSApplication.Main (System.String[] args) [0x00040] in <3e1eecfdb2534f518bea4f1a21972847>:0 
  at MatterControlMac.MainClass.Main (System.String[] args) [0x00012] in <f6b6c4b89de3427ab6b4c2f9608ae461>:0 
logout
Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.

[Process completed]

jlewin, 2019-09-21:
This issue is due to a regression that occured in: 

Added Export G-Code button to print menu (unlimitedbacon:master)
https://github.com/MatterHackers/MatterControl/pull/4569/files#diff-c36418805eb7f50619d283ee666c1cfdR203

The problem occurs because the `showInFolderAfterSave` widget is conditionally constructed on Windows only but is used without consideration in the call to the new `DoExport` function:
```csharp
DoExport(libraryItems, printer, activePlugin, centerOnBed, showInFolderAfterSave.Checked);
```

A simple fix can be achieved by using the null propagation operator to handle the null ref scenario encounter on Mac:
```csharp
DoExport(libraryItems, printer, activePlugin, centerOnBed, showInFolderAfterSave?.Checked == true);
```

I confirmed on the build server that problem originally repro'd and was resolved and worked as expected with the fix. I should have this update checked in next week and we can make a new build to test

jlewin, 2019-09-21:
One thing I failed to note is this bug should occur irrelevant of Mac OS version as we test for Windows and skip constructing `showInFolderAfterSave` when not found.

closed 2020-05-19 by larsbrubaker