MatterHackers/MatterControl #4680: Reload Failure status window pops up mid print

SMTDesigns, 2019-11-15:
Bug Report
============
**Steps to Reproduce**
1. nothing concrete actually

I have two win10 computers - one is controlling four printers and the'
other is controlling two. I start print jobs on all printers (all are two hour prints for example) and occasionally I get the Reload Failed message sometime during the print job (not always). Maybe on both systems, maybe just one.

**Expected Behavior**

Frankly, you should never get a Reload Failed message during printing. It should be something that happens when you select a bad profile for slicer, printer, etc. It should also not happen unless there is some type of permanent profile error. The message should really have more details as to what was trying to be loaded. Note: previous release would generate this message quite a lot when tweaking slicer profiles.

**Actual Behavior**
I get a Reload Failure message during printhing even though I haven't even touched the computer.

It does not cause an issue for the current printing job, but it seems to trash the Z offset for at least one printer. You also can not "fix" the value, it is stuck at 0.15 until you exit the program and reload it.

The offset I use is typically -0.1 to -0.3 depending on the printer (I do not have auto leveling and I do not use the leveling software in MatterControl).

**MatterControl Build Number** 2.19 latest master as of November 14, 2019.

(To find this, go to the MatterControl menu in the top left and click "About MatterControl")

**Operating System Version**

Win10 latest
**Printer Make/Model**
Anycubic i3 Mega-S

SMTDesigns, 2019-11-17:
I have to do more research on this. I realized that the two separate computers are interacting with each other through the MatterHackers account. When I setup the second computer, I selected an available printer from the choices I had already setup (I run six printers all identical except for their Z offset - four on one computer and two on the other). I would change a print profile from 35% fill to a solid on one computer and the setting would change on the other computer. I am guessing that the reload Failure occurred on the other system when I changed the fill density on the computer/printer I was working at. The print on the other computer finished and the Reload Failure happened.

From a hardware standpoint, a printer instance should refer to only one physical printer tied to the computer system it is on. The printer should be able to be "made" from a printer instance that exists on the same computer or a different computer. The instance is a physical port, physical computer, and a printer. There is no way to tell the difference between printers if they are all the same with the same firmware - the only difference will be they point of attachment - port and computer.

If anyone is experiencing temperature fluctuations between printers on different computers, it may be caused by the sync issue. I saw wide temp changes on the printers today after I started looking at the possible interaction between

SMTDesigns, 2019-11-17:
renamed the initial bug -

the second computer wasn't even setup for the cloud access. The only thing the two computers share is the same ethernet switch. I am in the process of redoing every printer except one to make sure they are totally isolated and will not disrupt each other. Even though I have no settings on any of the printers indicating a network connection, the two on my second computer will reflect even name changes back and forth with the first two printers on my first computer.

The Reload Failure was just a reaction to having MatterControl think there were two instances it needed to change on two different computers.

larsbrubaker, 2020-12-27:
Thanks for the feedback. The underlying issue with reload sometimes failing should be fixed.

closed 2020-12-27 by larsbrubaker