MatterHackers/MatterControl #4682: X11 Error: BadMatch (invalid parameter attributes) (crash)

jakimfett, 2019-11-27:
Bug Report
============
**Steps to Reproduce**
1. Upgrade to v2.19.10 on Elementary OS v5.0 Juno.  
2. (Attempt to) Start software with application tile, 
2b. Start software from command line.
2c. Momentary processing, some gtk/mono whinging.
3. X11 Error: BadMatch, crash with exit code 1.

**Expected Behavior**
Program loads and I can configure my 3d printer.

**Actual Behavior**
Program crashes.

**MatterControl Build Number**
Installed via `dpkg -i ./MatterControlSetup-2.19.10.10370.deb`, so assuming it's `10370`?

**Operating System Version**
Elementary OS 5.0 Juno

**Printer Make/Model**
Robo3D R1+.

**Console Log**
```
user@system >$ sudo dpkg -i ./MatterControlSetup-2.19.10.10370.deb 
[sudo] password for jakimfett:                            
Selecting previously unselected package mattercontrol.
(Reading database ... 266221 files and directories currently installed.)
Preparing to unpack .../MatterControlSetup-2.19.10.10370.deb ...
Unpacking mattercontrol (2.19.10.10367) ...
Setting up mattercontrol (2.19.10.10367) ...
Processing triggers for desktop-file-utils (0.23-1ubuntu3.18.04.2+elementary2~ubuntu5.0.1) ...
Processing triggers for gnome-menus (3.13.3-11ubuntu1.1) ...
Processing triggers for bamfdaemon (0.5.3+18.04.20180207.2-0ubuntu1) ...
Rebuilding /usr/share/applications/bamf-2.index...
Processing triggers for mime-support (3.60ubuntu1) ...

user@system >$ mattercontrol --version
Could not get XIM
Gtk-Message: 14:11:28.483: Failed to load module "pantheon-filechooser-module"

(mono:31518): Gtk-WARNING **: 14:11:28.498: Unable to locate theme engine in module_path: "pixmap",

(mono:31518): Gtk-WARNING **: 14:11:28.498: Unable to locate theme engine in module_path: "pixmap",

(mono:31518): Gtk-WARNING **: 14:11:28.498: Unable to locate theme engine in module_path: "pixmap",

(mono:31518): Gtk-WARNING **: 14:11:28.499: Unable to locate theme engine in module_path: "pixmap",
X11 Error encountered: 
  Error: BadMatch (invalid parameter attributes)
  Request:     154 (5)
  Resource ID: 0x7200014
  Serial:      264
  Hwnd:        Hwnd, Mapped:False ClientWindow:0x7200014, WholeWindow:0x7200013, Zombie=False, Parent:[Hwnd, Mapped:False ClientWindow:0x7200012, WholeWindow:0x7200011, Zombie=False, Parent:[<null>]]
  Control:     0  at System.Environment.get_StackTrace () [0x00000] in <285579f54af44a2ca048dad6be20e190>:0 
  at System.Windows.Forms.XplatUIX11.HandleError (System.IntPtr display, System.Windows.Forms.XErrorEvent& error_event) [0x00000] in <7346aaf818a24e898fc55461ebf690ad>:0 
  at OpenTK.Platform.X11.Glx.MakeCurrent (System.IntPtr , System.IntPtr , System.IntPtr ) [0x00000] in <d85857bc66524258b04ff0bdaeed67db>:0 
  at OpenTK.Platform.X11.Glx.MakeCurrent (System.IntPtr display, System.IntPtr drawable, OpenTK.ContextHandle context) [0x00000] in <d85857bc66524258b04ff0bdaeed67db>:0 
  at OpenTK.Platform.X11.X11GLContext.MakeCurrent (OpenTK.Platform.IWindowInfo window) [0x00000] in <d85857bc66524258b04ff0bdaeed67db>:0 
  at OpenTK.Graphics.GraphicsContext.MakeCurrent (OpenTK.Platform.IWindowInfo window) [0x00000] in <d85857bc66524258b04ff0bdaeed67db>:0 
  at OpenTK.GLControl.MakeCurrent () [0x00000] in <33e0a290e5404e8e8877ff9c93b1ec2f>:0 
  at OpenTK.GLControl.OnHandleCreated (System.EventArgs e) [0x00000] in <33e0a290e5404e8e8877ff9c93b1ec2f>:0 
  at System.Windows.Forms.Control.WmCreate (System.Windows.Forms.Message& m) [0x00000] in <7346aaf818a24e898fc55461ebf690ad>:0 
  at System.Windows.Forms.Control.WndProc (System.Windows.Forms.Message& m) [0x00000] in <7346aaf818a24e898fc55461ebf690ad>:0 
  at System.Windows.Forms.ScrollableControl.WndProc (System.Windows.Forms.Message& m) [0x00000] in <7346aaf818a24e898fc55461ebf690ad>:0 
  at System.Windows.Forms.ContainerControl.WndProc (System.Windows.Forms.Message& m) [0x00000] in <7346aaf818a24e898fc55461ebf690ad>:0 
  at System.Windows.Forms.UserControl.WndProc (System.Windows.Forms.Message& m) [0x00000] in <7346aaf818a24e898fc55461ebf690ad>:0 
  at System.Windows.Forms.Control+ControlWindowTarget.OnMessage (System.Windows.Forms.Message& m) [0x00000] in <7346aaf818a24e898fc55461ebf690ad>:0 
[... 32 lines cut ...]
  at System.Windows.Forms.NativeWindow.OnThreadException (System.Exception e) [0x00000] in <7346aaf818a24e898fc55461ebf690ad>:0 
  at System.Windows.Forms.NativeWindow.WndProc (System.IntPtr hWnd, System.Windows.Forms.Msg msg, System.IntPtr wParam, System.IntPtr lParam) [0x00196] in <7346aaf818a24e898fc55461ebf690ad>:0 
  at System.Windows.Forms.XplatUIX11.SendMessage (System.IntPtr hwnd, System.Windows.Forms.Msg message, System.IntPtr wParam, System.IntPtr lParam) [0x000d5] in <7346aaf818a24e898fc55461ebf690ad>:0 
  at System.Windows.Forms.XplatUIX11.CreateWindow (System.Windows.Forms.CreateParams cp) [0x004e5] in <7346aaf818a24e898fc55461ebf690ad>:0 
  at System.Windows.Forms.XplatUI.CreateWindow (System.Windows.Forms.CreateParams cp) [0x00000] in <7346aaf818a24e898fc55461ebf690ad>:0 
  at System.Windows.Forms.NativeWindow.CreateHandle (System.Windows.Forms.CreateParams cp) [0x00009] in <7346aaf818a24e898fc55461ebf690ad>:0 
  at System.Windows.Forms.Control.CreateHandle () [0x00031] in <7346aaf818a24e898fc55461ebf690ad>:0 
  at System.Windows.Forms.Control.CreateControl () [0x00039] in <7346aaf818a24e898fc55461ebf690ad>:0 
  at (wrapper remoting-invoke-with-check) System.Windows.Forms.Control.CreateControl()
  at System.Windows.Forms.Control.CreateControl () [0x00072] in <7346aaf818a24e898fc55461ebf690ad>:0 
  at System.Windows.Forms.Control.SetVisibleCore (System.Boolean value) [0x00057] in <7346aaf818a24e898fc55461ebf690ad>:0 
  at System.Windows.Forms.Form.SetVisibleCore (System.Boolean value) [0x00065] in <7346aaf818a24e898fc55461ebf690ad>:0 
  at MatterHackers.Agg.UI.WinformsSystemWindow.SetVisibleCore (System.Boolean value) [0x00009] in <712ffdb4c0144e8c88c0ce33a0b20977>:0 
  at System.Windows.Forms.Control.set_Visible (System.Boolean value) [0x00009] in <7346aaf818a24e898fc55461ebf690ad>:0 
  at System.Windows.Forms.Control.Show () [0x00000] in <7346aaf818a24e898fc55461ebf690ad>:0 
  at MatterHackers.Agg.UI.WinformsSystemWindow.Show () [0x000f8] in <712ffdb4c0144e8c88c0ce33a0b20977>:0 
  at MatterHackers.Agg.UI.WinformsSystemWindow.ShowSystemWindow (MatterHackers.Agg.UI.SystemWindow systemWindow) [0x00060] in <712ffdb4c0144e8c88c0ce33a0b20977>:0 
  at MatterHackers.Agg.UI.SingleWindowProvider.ShowSystemWindow (MatterHackers.Agg.UI.SystemWindow systemWindow) [0x00373] in <49bb6fd6740f4762a381356153fe2e02>:0 
  at MatterHackers.MatterControl.WinformsSingleWindowProvider.ShowSystemWindow (MatterHackers.Agg.UI.SystemWindow systemWindow) [0x00031] in <16b4831ddba24ef8b2ca5ca9a43af3cd>:0 
  at MatterHackers.Agg.UI.SystemWindow.ShowAsSystemWindow () [0x0002b] in <a5c2db3666d84398b1090e2f2dfee474>:0 
  at MatterHackers.MatterControl.Program.Main (System.String[] args) [0x0037f] in <e58f04b6857344df9781155f8ee2c0c8>:0 

user@system >$ echo $?
1
```

didli, 2019-12-08:
Same here with Fedora 31 xfce x64. Installed from unzipped .deb. Same errors.

jakimfett, 2019-12-16:
I'll go ahead and try to compile on my system once I [finish repairing my printer's hotend connector](https://www.reddit.com/r/3Dprinting/comments/e456se/burnt_connector_on_robo3d_r1_mainboard/), but I'd really much rather use the official pre-packaged version.

Is there some specific version of Mono that needs to be installed?  
Just by scanning the logfile, it looks like MatterControl is looking for "Microsoft.VisualBasic", which on 'Nix is going to be the `mono` package, right?

I've got version 6.6.0.161 installed, and I'm using the `.deb` file downloaded from the [MatterHackers store page](https://www.matterhackers.com/store/l/mattercontrol/sk/MKZGTDW6).

```
jakimfett@obsidian:~/Downloads$ which mono
/usr/bin/mono
jakimfett@obsidian:~/Downloads$ mono --version
Mono JIT compiler version 6.6.0.161 (tarball Tue Dec 10 10:36:32 UTC 2019)
Copyright (C) 2002-2014 Novell, Inc, Xamarin Inc and Contributors. www.mono-project.com
	TLS:           __thread
	SIGSEGV:       altstack
	Notifications: epoll
	Architecture:  amd64
	Disabled:      none
	Misc:          softdebug 
	Interpreter:   yes
	LLVM:          yes(610)
	Suspend:       hybrid
	GC:            sgen (concurrent by default)
```

SmashleyG, 2019-12-29:
exact same issue here on Xubuntu 19.10, downloaded .deb file from main site and have the following version of mono Mono JIT compiler version 6.6.0.161 (tarball Tue Dec 10 10:36:32 UTC 2019)

h54, 2019-12-31:
Same problem on Ubuntu 19.10 with .deb downloaded from MatterHackers site.

unlimitedbacon, 2020-01-02:
Can one of you try the instructions at the bottom of [this page](https://www.matterhackers.com/mattercontrol/support/linux) (the fix for low end GPUs) and see if that helps?

h54, 2020-01-02:
I tried it but it resulted in the same stackdump

jakimfett, 2020-01-02:
I do notice a graphics hiccup when between when the program is executed and the (moments later) crashdump.

That said, I'm running a relatively high-end graphics card (the [GTX 970 FTW+](https://www.videocardbenchmark.net/gpu.php?gpu=GeForce+GTX+970&id=2954), which is still in the top 100 graphics cards at nearly 5 years old), and setting `FSAASamples` to zero changed nothing (hiccup and crashdump still generated when running the `mattercontrol` command, run duration approximately three seconds, total).

didli, 2020-01-03:
Nvidia RTX 2060 founders edition, 440.44 Nvidia proprietary drivers. Coult it be that the convergent fault condition would be the nvidia proprietary drivers themselves, rather than the card ?

h54, 2020-01-04:
@didli  yes it would be interesting to see if anyone on AMD or Intel graphics is having this issue.

h54, 2020-01-23:
I installed 2.0 with only Intel Integrated graphics and it starts.  Initially, it's just a blank window but when I maximize it, I see and accept the license agreement.  After that, all is normal.

Fedora 31
UHD Graphics 620

jakimfett, 2020-01-24:
So, almost definitely an Nvidia upstream issue then, @h54 ?

h54, 2020-01-25:
@jakimfett I can't say definitively but I tried on two different systems with different Nvidia drivers (440.48.02 & 430.64) and the results are the same.  I don't have an AMD card to test with, unfortunately.

jakimfett, 2020-01-30:
> @h54 said:
> I can't say definitively but I tried on two different systems with different Nvidia drivers (440.48.02 & 430.64) and the results are the same. I don't have an AMD card to test with, unfortunately.

But we do know that Intel Integrated works.  
And those aren't from Nvidia, unless we're sorely mistaken.

h54, 2020-02-01:
I tested this morning with an AMD card and it started up without issue.

Pop_OS, kernel 5.3.0
RX-590

AngelQC, 2020-02-02:
I've got the same issue, archlinux w/nvidia GT530 - I know, an old card but as other software works (PrusaSlicer) I guess it should not matter...  
- kernel linux v5.4.15
- Driver nvidia-390xx 390.132-20 (because of old chipset)
- Library mono v6.4.0
Also, tried with and without the low end GPU edit.
Could not build mono48 (listed as compatible dependency, not sure if it would be preferable or not.)

docmerlin, 2020-02-05:
I get this problem with an NVIDIA 1080TI on archlinux

Brian245, 2020-03-04:
I'm getting the exact same error in the console log with Mint 18.3 running an Nvidia GeForce GTX 1060 6GB.

MechanizedMedic, 2020-10-19:
I am also getting this error running an AMD Radeon RX 570 card with the "amdgpu" open source driver on Arch Linux.

CAHbKA-IV, 2021-02-20:
You can edit /usr/lib/mattercontrol/appsettings.json to override this bug. Application crashes while 'Color' and 'Depth' parameters has different values. I set this values to '24', and program works fine.

didli, 2021-02-20:
Unbelievable, it works.

gpropf, 2021-10-04:
I can vouch for this fix as well. Set 'Color' and 'Depth' to 24. Program works fine.

ashleycawley, 2021-11-14:
Worked for me too! MatterControl wouldn't open for me on Zorin OS (Ubuntu based), but I got it working thanks to your advice by editing:
/usr/lib/mattercontrol/appsettings.json
And changing the color from 32 to 24 :+1:  thank you so much.

ashleycawley, 2021-11-15:
I contacted MatterHackers requesting that this user fix or work-around to the issue could be posted under the _**Know Issues**_ section of their documentation at https://matterhackers.github.io/MatterControl-Help/docs/linux

However I was directed to this GitHub, its unfortunate because this issue has been on-going for 2 years now and there are four independent issues lists on GitHub linked to to this issue, it seems like an issue that could potentially be fixed quite easily or at the very least the documentation updated so that users can help themselves to get up and running if the Linux software isn't going to be patched.

gpropf, 2021-11-16:
I'm glad my answer helped you. If I wasn't job hunting I'd fix it myself
but I can't seem to get the MC C# code to compile on my Linux machine and I
don't know enough about that environment to know why it won't work. This is
a low priority right now for me. The fix should be quite simple I would
think. Maybe somebody can make a binary without the bug and post it. - Greg