MatterHackers/MatterControl #4693: [Request] Object ID in gcode comments

Valinwolf, 2020-01-14:
Feature Request
===============

**Problem**
MatterSlice is my preferred slicer, however if I need to cancel _part_ of a print in OctoPrint, I'm left high and dry because MatterControl/MatterSlice does not add object ID comments and thus cannot be supported by OctoPrint plugin, [CancelObject](https://github.com/paukstelis/OctoPrint-Cancelobject).

**Solution**
Have each object automatically assigned an ID (even if just random UUIDs or something) and added as a comment above that object and an "end object" comment below each object on each layer. This will allow for @paukstelis to add support for MatterSlice.

This should probably be able to be disabled in settings and maybe even off by default for file size reasons.

paukstelis, 2020-01-14:
If this FR is implemented I would strongly recommend including M486 gcodes as well. This allows native object cancelling in Marlin firmware (2.0).

larsbrubaker, 2020-01-15:
I will give some thought to this and see about implementing. 

**Notes for developer:**

[M486 Documentation](https://reprap.org/wiki/G-code#M486:_Cancel_Object)
- [ ] Create bounding regions for mesh parts (start with AABB)
- [ ] On extrude check regions if diff than last emit exit and enter
- [ ] If non-region emit -1
- [ ] To start with overlapping parts may have ambiguous data

Sample Output from RepRap Wiki
```
M486 T12 ; Total of 12 objects (otherwise the firmware must count)
M486 S3  ; Indicate that the 4th object is starting now
M486 S-1 ; Indicate a non-object, purge tower, or other global feature
M486 P10 ; Cancel object with index 10 (the 11th object)
M486 U2  ; Un-cancel object with index 2 (the 3rd object)
M486 C   ; Cancel the current object (use with care!)
```