MatterHackers/MatterControl #4699: X11 Error/  Ubuntu 19.10

Marcus-Depew, 2020-02-08:
Bug Report
============
**Steps to Reproduce**
1. Ubuntu 19.10
2. MC.deb
3. Run from terminal 

**Actual Behavior**
MatterControl crashes 
Terminal dump:
Gtk not found (missing LD_LIBRARY_PATH to libgtk-x11-2.0.so.0?), using built-in colorscheme
Time to '': 158
X11 Error encountered:
Error: BadMatch (invalid parameter attributes)
Request: 155 (5)
Resource ID: 0x3A00016
Serial: 295
Hwnd: Hwnd, Mapped:False ClientWindow:0x3A00016, WholeWindow:0x3A00015, Zombie=False, Parent:[Hwnd, Mapped:False ClientWindow:0x3A00014, WholeWindow:0x3A00013, Zombie=False, Parent:[<null>]]
Control: 0 at System.Environment.get_StackTrace () [0x00000] in <c453bf849bb84e719dbec0475e99db95>:0
at System.Windows.Forms.XplatUIX11.HandleError (System.IntPtr display, System.Windows.Forms.XErrorEvent& error_event) [0x00000] in <1ad3b9c5e8484f3cb4aaaf843739b786>:0
at OpenTK.Platform.X11.Glx.MakeCurrent (System.IntPtr , System.IntPtr , System.IntPtr ) [0x00000] in <d85857bc66524258b04ff0bdaeed67db>:0
at OpenTK.Platform.X11.Glx.MakeCurrent (System.IntPtr display, System.IntPtr drawable, OpenTK.ContextHandle context) [0x00000] in <d85857bc66524258b04ff0bdaeed67db>:0
at OpenTK.Platform.X11.X11GLContext.MakeCurrent (OpenTK.Platform.IWindowInfo window) [0x00000] in <d85857bc66524258b04ff0bdaeed67db>:0
at OpenTK.Graphics.GraphicsContext.MakeCurrent (OpenTK.Platform.IWindowInfo window) [0x00000] in <d85857bc66524258b04ff0bdaeed67db>:0
at OpenTK.GLControl.MakeCurrent () [0x00000] in <33e0a290e5404e8e8877ff9c93b1ec2f>:0
at OpenTK.GLControl.OnHandleCreated (System.EventArgs e) [0x00000] in <33e0a290e5404e8e8877ff9c93b1ec2f>:0
at System.Windows.Forms.Control.WmCreate (System.Windows.Forms.Message& m) [0x00000] in <1ad3b9c5e8484f3cb4aaaf843739b786>:0
at System.Windows.Forms.Control.WndProc (System.Windows.Forms.Message& m) [0x00000] in <1ad3b9c5e8484f3cb4aaaf843739b786>:0
at System.Windows.Forms.ScrollableControl.WndProc (System.Windows.Forms.Message& m) [0x00000] in <1ad3b9c5e8484f3cb4aaaf843739b786>:0
at System.Windows.Forms.ContainerControl.WndProc (System.Windows.Forms.Message& m) [0x00000] in <1ad3b9c5e8484f3cb4aaaf843739b786>:0
at System.Windows.Forms.UserControl.WndProc (System.Windows.Forms.Message& m) [0x00000] in <1ad3b9c5e8484f3cb4aaaf843739b786>:0
at System.Windows.Forms.Control+ControlWindowTarget.OnMessage (System.Windows.Forms.Message& m) [0x00000] in <1ad3b9c5e8484f3cb4aaaf843739b786>:0
at System.Windows.Forms.Control+ControlNativeWindow.WndProc (System.Windows.Forms.Message& m) [0x00000] in <1ad3b9c5e8484f3cb4aaaf843739b786>:0
at System.Windows.Forms.NativeWindow.WndProc (System.IntPtr hWnd, System.Windows.Forms.Msg msg, System.IntPtr wParam, System.IntPtr lParam) [0x00000] in <1ad3b9c5e8484f3cb4aaaf843739b786>:0
at System.Windows.Forms.XplatUIX11.SendMessage (System.IntPtr hwnd, System.Windows.Forms.Msg message, System.IntPtr wParam, System.IntPtr lParam) [0x00000] in <1ad3b9c5e8484f3cb4aaaf843739b786>:0
at System.Windows.Forms.XplatUIX11.CreateWindow (System.Windows.Forms.CreateParams cp) [0x00000] in <1ad3b9c5e8484f3cb4aaaf843739b786>:0
at System.Windows.Forms.XplatUI.CreateWindow (System.Windows.Forms.CreateParams cp) [0x00000] in <1ad3b9c5e8484f3cb4aaaf843739b786>:0
at System.Windows.Forms.NativeWindow.CreateHandle (System.Windows.Forms.CreateParams cp) [0x00000] in <1ad3b9c5e8484f3cb4aaaf843739b786>:0
at System.Windows.Forms.Control.CreateHandle () [0x00000] in <1ad3b9c5e8484f3cb4aaaf843739b786>:0
at System.Windows.Forms.Control.CreateControl () [0x00000] in <1ad3b9c5e8484f3cb4aaaf843739b786>:0
at System.Windows.Forms.Control.CreateControl () [0x00000] in <1ad3b9c5e8484f3cb4aaaf843739b786>:0
at System.Windows.Forms.Control.SetVisibleCore (System.Boolean value) [0x00000] in <1ad3b9c5e8484f3cb4aaaf843739b786>:0
at System.Windows.Forms.Form.SetVisibleCore (System.Boolean value) [0x00000] in <1ad3b9c5e8484f3cb4aaaf843739b786>:0
at MatterHackers.Agg.UI.WinformsSystemWindow.SetVisibleCore (System.Boolean value) [0x00000] in <712ffdb4c0144e8c88c0ce33a0b20977>:0
at System.Windows.Forms.Control.set_Visible (System.Boolean value) [0x00000] in <1ad3b9c5e8484f3cb4aaaf843739b786>:0
at System.Windows.Forms.Control.Show () [0x00000] in <1ad3b9c5e8484f3cb4aaaf843739b786>:0
at MatterHackers.Agg.UI.WinformsSystemWindow.Show () [0x00000] in <712ffdb4c0144e8c88c0ce33a0b20977>:0
at MatterHackers.Agg.UI.WinformsSystemWindow.ShowSystemWindow (MatterHackers.Agg.UI.SystemWindow systemWindow) [0x00000] in <712ffdb4c0144e8c88c0ce33a0b20977>:0
at MatterHackers.Agg.UI.SingleWindowProvider.ShowSystemWindow (MatterHackers.Agg.UI.SystemWindow systemWindow) [0x00000] in <49bb6fd6740f4762a381356153fe2e02>:0
at MatterHackers.MatterControl.WinformsSingleWindowProvider.ShowSystemWindow (MatterHackers.Agg.UI.SystemWindow systemWindow) [0x00000] in <16b4831ddba24ef8b2ca5ca9a43af3cd>:0
at MatterHackers.Agg.UI.SystemWindow.ShowAsSystemWindow () [0x00000] in <a5c2db3666d84398b1090e2f2dfee474>:0
at MatterHackers.MatterControl.Program.Main (System.String[] args) [0x00000] in <e58f04b6857344df9781155f8ee2c0c8>:0

exception inside UnhandledException handler: (null) assembly:/usr/lib/mono/4.5/mscorlib.dll type:FileNotFoundException member:(null)

[ERROR] FATAL UNHANDLED EXCEPTION: System.IO.FileNotFoundException: Could not load file or assembly 'Microsoft.VisualBasic, Version=10.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a' or one of its dependencies.
File name: 'Microsoft.VisualBasic, Version=10.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a'
at Mindscape.Raygun4Net.RaygunMessageBuilder.SetEnvironmentDetails () [0x0000b] in <f91f2b249ee243acbec1dd6bfd729d19>:0
at Mindscape.Raygun4Net.RaygunClient.BuildMessage (System.Exception exception, System.Collections.Generic.IList`1[T] tags, System.Collections.IDictionary userCustomData, Mindscape.Raygun4Net.Messages.RaygunIdentifierMessage userInfoMessage, System.Nullable`1[T] currentTime) [0x00024] in <f91f2b249ee243acbec1dd6bfd729d19>:0
at Mindscape.Raygun4Net.RaygunClient.StripAndSend (System.Exception exception, System.Collections.Generic.IList`1[T] tags, System.Collections.IDictionary userCustomData, Mindscape.Raygun4Net.Messages.RaygunIdentifierMessage userInfo, System.Nullable`1[T] currentTime) [0x00016] in <f91f2b249ee243acbec1dd6bfd729d19>:0
at Mindscape.Raygun4Net.RaygunClient.Send (System.Exception exception, System.Collections.Generic.IList`1[T] tags, System.Collections.IDictionary userCustomData, Mindscape.Raygun4Net.Messages.RaygunIdentifierMessage userInfo) [0x00023] in <f91f2b249ee243acbec1dd6bfd729d19>:0
at Mindscape.Raygun4Net.RaygunClient.Send (System.Exception exception) [0x00000] in <f91f2b249ee243acbec1dd6bfd729d19>:0
at MatterHackers.MatterControl.Program+<>c.<Main>b__6_2 (System.Object s, System.Threading.ThreadExceptionEventArgs e) [0x0001c] in <e58f04b6857344df9781155f8ee2c0c8>:0
at System.Windows.Forms.Application.OnThreadException (System.Exception t) [0x00037] in <1ad3b9c5e8484f3cb4aaaf843739b786>:0
at System.Windows.Forms.NativeWindow.OnThreadException (System.Exception e) [0x00000] in <1ad3b9c5e8484f3cb4aaaf843739b786>:0
at System.Windows.Forms.NativeWindow.WndProc (System.IntPtr hWnd, System.Windows.Forms.Msg msg, System.IntPtr wParam, System.IntPtr lParam) [0x00196] in <1ad3b9c5e8484f3cb4aaaf843739b786>:0
at System.Windows.Forms.XplatUIX11.SendMessage (System.IntPtr hwnd, System.Windows.Forms.Msg message, System.IntPtr wParam, System.IntPtr lParam) [0x000d5] in <1ad3b9c5e8484f3cb4aaaf843739b786>:0
at System.Windows.Forms.XplatUIX11.CreateWindow (System.Windows.Forms.CreateParams cp) [0x004e5] in <1ad3b9c5e8484f3cb4aaaf843739b786>:0
at System.Windows.Forms.XplatUI.CreateWindow (System.Windows.Forms.CreateParams cp) [0x00000] in <1ad3b9c5e8484f3cb4aaaf843739b786>:0
at System.Windows.Forms.NativeWindow.CreateHandle (System.Windows.Forms.CreateParams cp) [0x00009] in <1ad3b9c5e8484f3cb4aaaf843739b786>:0
at System.Windows.Forms.Control.CreateHandle () [0x00031] in <1ad3b9c5e8484f3cb4aaaf843739b786>:0
at System.Windows.Forms.Control.CreateControl () [0x00039] in <1ad3b9c5e8484f3cb4aaaf843739b786>:0
at (wrapper remoting-invoke-with-check) System.Windows.Forms.Control.CreateControl()
at System.Windows.Forms.Control.CreateControl () [0x00072] in <1ad3b9c5e8484f3cb4aaaf843739b786>:0
at System.Windows.Forms.Control.SetVisibleCore (System.Boolean value) [0x00057] in <1ad3b9c5e8484f3cb4aaaf843739b786>:0
at System.Windows.Forms.Form.SetVisibleCore (System.Boolean value) [0x00065] in <1ad3b9c5e8484f3cb4aaaf843739b786>:0
[... cut ...]
at MatterHackers.Agg.UI.SingleWindowProvider.ShowSystemWindow (MatterHackers.Agg.UI.SystemWindow systemWindow) [0x00373] in <49bb6fd6740f4762a381356153fe2e02>:0
at MatterHackers.MatterControl.WinformsSingleWindowProvider.ShowSystemWindow (MatterHackers.Agg.UI.SystemWindow systemWindow) [0x00031] in <16b4831ddba24ef8b2ca5ca9a43af3cd>:0
at MatterHackers.Agg.UI.SystemWindow.ShowAsSystemWindow () [0x0002b] in <a5c2db3666d84398b1090e2f2dfee474>:0
at MatterHackers.MatterControl.Program.Main (System.String[] args) [0x0037f] in <e58f04b6857344df9781155f8ee2c0c8>:0

**MatterControl Build Number**
2.19.10.10367
**Operating System Version**
Ubuntu 19.10

pezr, 2020-04-12:
I know Archlinux distro is not officialy supported. I've installed from the AUR:
https://aur.archlinux.org/packages/mattercontrol/

I'm having the same error:
exception inside UnhandledException handler: (null) assembly:/usr/lib/mono/4.5/mscorlib.dll type:FileNotFoundException member:(null)

Altough, I have mono installed and the file /usr/lib/mono/4.5/mscorlib.dll in place.

Greetings

MatterControl Build Number
2.19.10.10370

pezr, 2020-04-12:
Ok, this is related to this one 4682 and appears to be a NVIDIA drivers issue.