MatterHackers/MatterControl #4723: Printer with a negative Probe offset

Bob590, 2020-04-09:
Bug Report
============
**Steps to Reproduce**
1.  You need to place a Probe on the X left of the noozle so that the probe offset is negative (in my case it is -40).
2.  The probe Y offset is zero in my case but could be negative as well. You also need to take into consideratrion this possibility.
3.  Try Bed Leveling with a 10x10 grid
I have tried the 3x3 and 5x5 and they succeed but I wish a more detailed leveling. ;-)

**Expected Behavior**
The printer has been tested to succeed on all points inside the bed that is 200x200 using the G30 command after positioning with G1 command.
The bed leveling should succeed. Alls point have been sampled (G1, G30 sequence).

**Actual Behavior**
The bed leveling completes but request another turn, again and again, because it doesn't pass validation: some sampled points are the same. BTW: I had to discover this by checking the source code. I suggest that a proper message is shown to the user or he will not understand WHY the program asks to repeat again and again!

**MatterControl Build Number**
(To find this, go to the MatterControl menu in the top left and click "About MatterControl")
2.20.4.10449

**Operating System Version**
Windows 7 and Windows 10 (I have tested both)

**Printer Make/Model**
Anycube i3 equivalent to Prusa mk2

I have downloaded the source code and I see that the function 
public override IEnumerable<Vector2> GetPrintLevelPositionToSample()
doesn't take into account probe offsets.
Thus the first sample point is x20 y19.
Considering that my probe x offset is -40 then the first point should be 40. In particular I expect X 
 values similar to these: 40, 57.77, 75.55, 93.31, 111.09, 128.86, 146.62, 164.39, 182.17, 199.94
or near these values.
Instead the X sampled points are: 20, 37.77, 55.55, 73.33, 91.11, 108.88, 126.66, 144.44, 160, 160
This lead to repeat some sampled points and thus the validation FAILS!

I could modify myself the current scource code but, of course, I prefer that you give a different solution or that you fix the code for EVERYONE.

Thank you,
Bob

Bob590, 2020-04-11:
Just for being of some help here is my current solution to this problem. NOTE I only validated it for the 10x10 Mesh (the one I use) so you need to review all other options... ;-)

Change 1:
       //Bob: modified to take into account Probe offsets
        public override IEnumerable<Vector2> GetPrintLevelPositionToSample()
		{
			Vector2 bedSize = printer.Settings.GetValue<Vector2>(SettingsKey.bed_size);
			Vector2 printCenter = printer.Settings.GetValue<Vector2>(SettingsKey.print_center);

			if (printer.Settings.GetValue<BedShape>(SettingsKey.bed_shape) == BedShape.Circular)
			{
				// reduce the bed size by the ratio of the radius (square root of 2) so that the sample positions will fit on a circular bed
				bedSize *= 1.0 / Math.Sqrt(2);
			}

            //double halfXSize = (bedSize.X / 2) * .8;
            //double xStep = (halfXSize * 2) / (gridWidth - 1);
            //double halfYSize = (bedSize.Y / 2) * .8;
            //double yStep = (halfYSize * 2) / (gridHeight - 1);

            const double cMargin = 10;
            var probeOffset = printer.Settings.GetValue<Vector3>(SettingsKey.probe_offset);
            var x_start = probeOffset.X < 0 ? -probeOffset.X : 0;
            x_start += cMargin;
            var x_end = probeOffset.X < 0 ? bedSize.X : bedSize.X - probeOffset.X;
            double xStep = (x_end - x_start - cMargin) / (gridWidth - 1);
            var y_start = probeOffset.Y < 0 ? -probeOffset.Y : 0;
            y_start += cMargin;
            var y_end = probeOffset.Y < 0 ? bedSize.Y : bedSize.Y - probeOffset.Y;
            double yStep = (y_end - y_start - cMargin) / (gridHeight - 1);

            for (int y = 0; y < gridHeight; y++)
			{
				// make it such that every other line is printed from right to left
                for (int x = 0; x < gridWidth; x++)
				{
					int dirX = x;
					if ((y % 2) == 1)
					{
						dirX = (gridWidth - 1) - x;
					}

					var samplePosition = new Vector2();
                    //samplePosition.X = printCenter.X - halfXSize + (dirX * xStep);
                    samplePosition.X = x_start + (dirX * xStep);
                    //samplePosition.Y = printCenter.Y - halfYSize + (y * yStep);
                    samplePosition.Y = y_start + (y * yStep);

                    yield return samplePosition;
				}
			}
		}

The basic idea is create probe points in places where the probe can actually works!

Change 2:
in public static Vector2 EnsureInPrintBounds(PrinterConfig printer, Vector2 probePosition)

			if (printer.Settings.Helpers.UseZProbe())
			{
				var probeOffset2D = new Vector2(printer.Settings.GetValue<Vector3>(SettingsKey.probe_offset));
				//var actualNozzlePosition = probePosition - probeOffset2D;
                /*Bob: */
                Vector2 actualNozzlePosition;
                if(probeOffset2D.X >= 0 )
                    actualNozzlePosition.X = probePosition.X - probeOffset2D.X;
                else
                    actualNozzlePosition.X = probePosition.X + probeOffset2D.X;
                if (probeOffset2D.Y >= 0)
                    actualNozzlePosition.Y = probePosition.Y - probeOffset2D.Y;
                else
                    actualNozzlePosition.Y = probePosition.Y + probeOffset2D.Y;
                /**/
                // clamp this to the bed bounds
                Vector2 bedSize = printer.Settings.GetValue<Vector2>(SettingsKey.bed_size);
				Vector2 printCenter = printer.Settings.GetValue<Vector2>(SettingsKey.print_center);
				var bedBounds = new RectangleDouble(printCenter - bedSize / 2, printCenter + bedSize / 2);
				Vector2 adjustedPosition = bedBounds.Clamp(actualNozzlePosition);

                //StyledMessageBox.ShowMessageBox(string.Format("probePosition X={0}, Y={1}\n\nadjustedPosition X={2}, Y={3}", probePosition.X, probePosition.Y, adjustedPosition.X, adjustedPosition.Y), "Debug" );

                // and push it back into the probePosition
                //Bob: probePosition = adjustedPosition + probeOffset2D;
			}

change 3:
in private void StartSampling(int probeIndex, Vector3 probePosition)
just comment this line:

//Bob: adjustedProbePosition -= probeOffset;

The adjustments have been commented because as the probe points are correct when generated there is no more need to FIX them.

I hope this could help and speed up your official fixing of this problem!

Thank you,
Bob

larsbrubaker, 2020-04-16:
Thanks for the feedback and the code. There are a few more subtleties to what all is happening, but you have captured the problem. We will work on making this better.

Bob590, 2020-04-21:
I hope you will release soon a new version that will cover this problem.
Meanwhile I'm trying to use my modified version even if it not perfect.

BTW: I wish to ask if the Z offset can be modified after you have performed the Z leveling in order to compensate tolerances in Z. I mean that initially you have problems to precisely calibrate the first layer adherence and thus you need to sligtly modify the Z in order to have the PLA build the first layer precisely. Example: suppose that my probe is 1.1mm upper respect to the noozle, and that the leveling procedure that calibrate Z offset stores in the printer settings Z offset = 0.4mm. At this point if I find that my first layer is still OVER my bed, can I adjust 0.4 to 0.3 or 0.2 to compensate this? Of course I need to slice again after any change....
Thank you very much for your support and great work!

larsbrubaker, 2021-11-30:
This should be fixed. 

**If it is not** 
- Please re-open this issue
- Please export, zip and add your profile to this issue so we can investigate exactly what is not yet working

closed 2021-11-30 by larsbrubaker