MatterHackers/MatterControl #4726: Open a file

adam57jerome, 2020-04-12:
Bug Report
============
**Steps to Reproduce**
1. Open a file
2. 
3. 

**Expected Behavior**
A window opens and I select my * .stl file for example
**Actual Behavior**
nothing happens, no window, I can't select a file.
I have to use the cloud

**MatterControl Build Number**
(To find this, go to the MatterControl menu in the top left and click "About MatterControl")
2.19.10.10370

**Operating System Version**
linux ubuntu 19.10
**Printer Make/Model**
Anet A6 merlin 1.9.1

Marcus-Depew, 2020-04-24:
Try these steps for installing Mono (link below), but you can also drag and drop an STL onto the print bed. 

https://www.matterhackers.com/mattercontrol/support/linux

eguane, 2020-04-24:
Like the bug report, ubuntu 19.10 and MatterControl 2.19.10.10370
Not possible to change image, no path windows to select the file.
Mono updated like you recommendation but no effect.
Rgds

Marcus-Depew, 2020-04-26:
You are right, that MC Linux support page is missing a step and will be amended soon, in the mean time checkout the official Mono Development install page:
  https://www.monodevelop.com/download/

The step that is missing is: sudo apt-get install monodevelop
I was able to recreate symptoms you described, after I double checking that monodevelop was installed fixed it :) However please let me know if the problem persists.

inferont, 2020-05-29:
I have this same issue in Ubuntu 20.04. I've installed mono-devel from the mono repository. This issue appears to affect a few other areas too:
* Exporting gcode or other files
* Saving a file (The save button is always greyed out)
* Clicking on links in the Store

MarcBeal, 2020-06-05:
> You are right, that MC Linux support page is missing a step and will be amended soon, in the mean time checkout the official Mono Development install page:
> https://www.monodevelop.com/download/
>
> The [...]

Thanks Marcus-Depew! This issue appeared today randomly for me (not sure if it was related to any of my recent updates). I was unable to export anything. Installing monodevelop fixed the export issue.

lsnadon, 2021-12-10:
Old thread - but this fixed it for me - I loaded everything mono related.  It would be nice to know which package fixed it, I know, but I got sick of trying this and that.  sudo apt-get install mono-* solved my issue.