MatterHackers/MatterControl #4735: OpenTK issue

matthewcook10, 2020-04-28:
Tried installed on Arch from the AUR. Everything seems to work fine until it launches.

OpenTK.Graphics.GraphicsModeException: Requested GraphicsMode not available.
  at OpenTK.Platform.X11.X11GraphicsMode.SelectGraphicsMode (OpenTK.Graphics.ColorFormat color, System.Int32 depth, System.Int32 stencil, System.Int32 samples, OpenTK.Graphics.ColorFormat accum, System.Int32 buffers, System.Boolean stereo) [0x00054] in <d85857bc66524258b04ff0bdaeed67db>:0 
  at OpenTK.Graphics.GraphicsMode.LazySelectGraphicsMode () [0x00010] in <d85857bc66524258b04ff0bdaeed67db>:0 
  at OpenTK.Graphics.GraphicsMode.get_Index () [0x00000] in <d85857bc66524258b04ff0bdaeed67db>:0 
  at OpenTK.X11GLControl..ctor (OpenTK.Graphics.GraphicsMode mode, System.Windows.Forms.Control control) [0x00022] in <33e0a290e5404e8e8877ff9c93b1ec2f>:0 
  at OpenTK.GLControlFactory.CreateGLControl (OpenTK.Graphics.GraphicsMode mode, System.Windows.Forms.Control control) [0x00041] in <33e0a290e5404e8e8877ff9c93b1ec2f>:0 
  at OpenTK.GLControl.OnHandleCreated (System.EventArgs e) [0x00046] in <33e0a290e5404e8e8877ff9c93b1ec2f>:0 
  at System.Windows.Forms.Control.WmCreate (System.Windows.Forms.Message& m) [0x00000] in <5ad3d0bd1cd24243af30eb325f259ec8>:0 
  at System.Windows.Forms.Control.WndProc (System.Windows.Forms.Message& m) [0x0021c] in <5ad3d0bd1cd24243af30eb325f259ec8>:0 
  at System.Windows.Forms.ScrollableControl.WndProc (System.Windows.Forms.Message& m) [0x00000] in <5ad3d0bd1cd24243af30eb325f259ec8>:0 
  at System.Windows.Forms.ContainerControl.WndProc (System.Windows.Forms.Message& m) [0x00027] in <5ad3d0bd1cd24243af30eb325f259ec8>:0 
  at System.Windows.Forms.UserControl.WndProc (System.Windows.Forms.Message& m) [0x00025] in <5ad3d0bd1cd24243af30eb325f259ec8>:0 
  at System.Windows.Forms.Control+ControlWindowTarget.OnMessage (System.Windows.Forms.Message& m) [0x00000] in <5ad3d0bd1cd24243af30eb325f259ec8>:0 
  at System.Windows.Forms.Control+ControlNativeWindow.WndProc (System.Windows.Forms.Message& m) [0x0000b] in <5ad3d0bd1cd24243af30eb325f259ec8>:0 
  at System.Windows.Forms.NativeWindow.WndProc (System.IntPtr hWnd, System.Windows.Forms.Msg msg, System.IntPtr wParam, System.IntPtr lParam) [0x00085] in <5ad3d0bd1cd24243af30eb325f259ec8>:0 

The window instantly crashes.  Any help/advice would be greatly appreciated!

matthewcook10, 2020-04-28:
Forgot to mention, I get this same error when building from source as well so its not an AUR thing. I don't know if it comes down to a system issue or code issue. Any help please.

dther, 2022-05-05:
I figured this out! I was getting the same error, so I made the following changes in `/usr/lib/mattercontrol/appsettings.json`:

```
...
    "GraphicsMode": {
      "Color": 24,
      "Depth": 24,
      "FSAASamples": 1
    },
...
```
`FSAASamples` was originally 8, and apparently `color` and `depth` being different from each other apparently causes problems with X. I suspect this problem is due to Intel Integrated Graphics not having the necessary functions to handle higher FSAASamples.