MatterHackers/MatterControl #4743: Saved files can not open correct on other systems

AzeOfChaze, 2020-05-09:
Bug Report
============
**Steps to Reproduce**
1. Save a file containing some designed objects/shapes, to be opened on another system
2. Open the file at the other computer

**Expected Behavior**
Identical file for further working process 

**Actual Behavior**
All the shapes are replaced by questions marks, as if the system don't recognize it.
Have tried on 4 different systems, both saving/open files in between

**MatterControl Build Number**
(To find this, go to the MatterControl menu in the top left and click "About MatterControl")
2.20.4.10449

**Operating System Version**
Windows 10 64 bit

**Printer Make/Model**
Prusa i3 mk3s
[image]

robotsBill, 2020-05-14:
This also occurs when opening an .mcx file created on a mac under 10.15.4
    (Mattercontrol v2.20.4.10455)
geometry comes in as question shapes on a windows 10 PC 
(Matterconrtrol v2.20.449)

Going from the PC to the Mac works!

AzeOfChaze, 2020-05-15:
robotsBill:
Thank you very much for your reply. 

So this means that it is impossible to save files on Windows systems, and then open it on another Windows? 
This makes no sense to me and must be a bug?!

larsbrubaker, 2020-05-15:
If you save to your cloud library you can open that on any system. The issue is that on a local .mcx file it is referencing the .stl files directly on your had drive, not including copies of them within it. When you transport this file to another machine it does not have the .stl files.

We plan in the future to make it possible to export to a .mcx file that contains all of the assets and will be transportable.

AzeOfChaze, 2020-05-15:
larsbrubaker: Thank you SO much for your reply. I was actually hoping that you would reply, as I have seen a lot of posts from you, and you seem to be the right person for this matter! 

Ok, that makes sense to me. But the problem is, I don't get the option saving to the cloud?
[image]

robotsBill, 2020-05-15:
This is a Mac to PC issue. (My home is Mac & my work is Windows)
I can open the file on the Mac, but each solid shape is a 3D question mark, proportioned like the original.   Colors seem to be correct.  Cylindrical objects have square edges (like the cube objects, just in the shape of a question mark!

Geometry generated on the PC open fine on the Mac.

I could use Virtual Pc, but that has its own baggage!

robotsBill, 2020-05-15:
All of the geometry was created in session using the matter control primitives.  I don’t understand what .stl’s would be involved.  I’m glad that you have a future solution.

Bill

larsbrubaker, 2020-05-15:
Thank you for the clarification. Getting things to rebuild that are made out of matter control primitives will actually happen sooner than the ability to export an entire set. That is something that we are working on very soon.

AzeOfChaze, 2020-05-15:
larsbruker: So that means that it will be possible to save as an .mcx to my dropbox, and then open the file from another location?
Looking VERY forward to this.

In the meantime, can you tell me how to export the files to the cloud maybe? I have got a new laptop and would like to transfer the files to it for further designing!

larsbrubaker, 2020-05-15:
Here is a video of saving to the could folder. You need to be logged in to save to your cloud folder.

[image]

BTW, there was already a cube on the bed when I started (off camera) so the final open is right.

AzeOfChaze, 2020-05-16:
lars: You're the man!!! Thank you SO much. Now it all starts to look bright.

How much storage space do I have in the MC Cloud?

closed 2020-05-16 by AzeOfChaze