MatterHackers/MatterControl #4744: MatterControl 2.0 (2.20.4.10449) - G-Code export message "Disabled: Setup Needs to be Run"

hnolsoe, 2020-05-09:
Bug Report
============
**Steps to Reproduce**
1. Import .STL
2. Slice to produce G-Code
3. In bed menu select item "export"

**Expected Behavior**
In previous verion 1.7 this results in a pop-up window where "Export selection to:" option "Machine File (G-Code) is available.

**Actual Behavior**
In current verion 2.0 this results in a pop-up window where "Export selection to:" option "Machine File (G-Code) is disabled with the following message:
"Disabled: Setup Needs to be Run"

**MatterControl Build Number**
build 2.20.4.10449

**Operating System Version**
Windows 10 Enterprise 64-bit

**Printer Make/Model**
Creality CR3040S - but that is irrelevant because it is not connected to the printer directly.

larsbrubaker, 2020-05-16:
Tested and working with profile Creality CR-10 S. Will be in the text update (due out near the beginning of June).

If this is not working as expected after the next release, please compare your profile with the shipping CR-10 S pofile.

Thank you for the report.

closed 2020-05-16 by larsbrubaker

hnolsoe, 2020-05-17:
I did indeed use the CR-10 S profile while I had the issue, but I did never connect MatterControl to the printer directly (via USB). 
However a few days ago I decided to connect directly to the printer and the "Setup Wizard" popped up. I chose to go through the wizard steps without changing anything (since the printer is setup fine) and after that my g-code export issue is fixed.