MatterHackers/MatterControl #4753: Export GCode controls fail silently

gvcastellano, 2020-05-20:
Bug Report
============
**Steps to Reproduce**
1. Connect printer (same issue arises if no printer connected)
2. Add small project to bed from .stl file
3. From the Print menu, hit "Export", or
4. Right-click on the bed, select Export, select G-Code, hit Export button

**Expected Behavior**
GCode file is available somewhere in home filesystem.

**Actual Behavior**
Export selection in print menu appears to have no effect;
Export button in Export File dialog closes dialog but no other effect.

**MatterControl Build Number**
(To find this, go to the MatterControl menu in the top left and click "About MatterControl")
2.19.10.10370

**Operating System Version**
Ubuntu 20.04 LTS Gnome-MATEexport

**Printer Make/Model**
Robo3D R1+

inferont, 2020-05-29:
I also have this issue. It might be related to a similar issue I have in #4726

gvcastellano, 2020-05-29:
Thx but if that's about installing mono-devel, I already have it:

ii  mono-devel                                                  6.8.0.105+dfsg-2                  all          Mono development tools

inferont, 2020-06-01:
I have all the mono packages installed with mono-devel too. I still am having these issues. My comment was for whoever was working on this. These issues may have the same cause.

gvcastellano, 2020-07-04:
I have figured out a workaround.  In MatterControl, load your model onto the bed, then hit the "Slice" tab.  Then from the command line, locate the .gcode files created within the past day:

gvc@spock:~/Dropbox/midi$ find ~ -name "*.gcode" -mtime -1 -ls
  3152444    228 -rw-rw-r--   1 gvc      gvc        233013 Jul  4 14:07 /home/gvc/.local/share/MatterControl/data/temp/gcode/0E8B2435B6A6047EA7A3FD9A25797EF18429958A_5447122326331654270.gcode
  3152423    244 -rw-rw-r--   1 gvc      gvc        246768 Jul  4 13:28 /home/gvc/.local/share/MatterControl/data/temp/gcode/0E8B2435B6A6047EA7A3FD9A25797EF18429958A_11974058689766276895.gcode

Once you've located the directory where the .gcode is written there are plenty other ways to locate the latest files, for example:

gvc@spock:~/Dropbox/midi$ ls -ltr /home/gvc/.local/share/MatterControl/data/temp/gcode/ | tail -5
-rw-rw-r-- 1 gvc gvc 12768520 Jun 21 20:56 BDF6C54D003C8E4CB3A046CFBA103528EAF3DFA0_3457962567905867514.gcode
-rw-rw-r-- 1 gvc gvc 12564287 Jun 22 11:10 BDF6C54D003C8E4CB3A046CFBA103528EAF3DFA0_11974058689766276895.gcode
-rw-rw-r-- 1 gvc gvc 12441485 Jun 22 11:19 BF6B6391D279C345197A16A739DAE4BD3141754E_11974058689766276895.gcode
-rw-rw-r-- 1 gvc gvc     2958 Jul  4 13:28 config_11974058689766276895.ini
-rw-rw-r-- 1 gvc gvc   246768 Jul  4 13:28 0E8B2435B6A6047EA7A3FD9A25797EF18429958A_11974058689766276895.gcode

The filenames are unintelligible but I just pay attention to the datestamps.  To print from the SD card, copy to it with the name "auto0.g", unplug the printer's USB from any host computer and power it up.  Since everything I need to know is in the .gcode file (in particular the print duration) I am good to go.

I had known how to find the .gcode files for a while but for some reason did not believe they were complete.  I think I was being tripped up by the need to unplug the USB cable, which otherwise appears to inhibit the auto0.g function.

It would still be nice to have the Mono file classes (as I assume that's what broken) working so I have more options for importing .stls and don't have to go fishing for the .gcode but all in the life of a hacker, most days I commit 6 worse kludges before breakfast.

code-to-joy, 2020-11-23:
I have this exact same issue. It practically makes it impossible to use MC for printing. This must affect a great deal of users? Is there an a earlier version that does not experience this issue that one can downgrade to?

larsbrubaker, 2020-11-23:
This report is specific to linux (which we are hopping to have a new version for soon). If you are seeing this on another platform, please file a separate bug as that would represent a new issue.

code-to-joy, 2020-11-23:
I'm on Linux too, yes. Elementary OS, which currently is 18.04.1-Ubuntu. I can't export stl-files either, same issue. 

Also, the workaround provided by @gvcastellano above doesn't work for me, since the gcode files written to the temp folder doesn't contain the actual print instructions, but rather the print job config/details - I only get a ~4 kB file, when it should be in the MB's. 

The last line in all my exported files is "numberOfBrimLoops = 13" - could that be a config issue that causes this?

snarlysodboxer, 2020-12-21:
I'm having the same problem on Ubuntu 20.04. Nothing happens using either of the two export dialogs (under the 'Print' or 'Bed' buttons.) Additionally, if I go to Settings -> Application Storage -> Open Folder, nothing happens.

I installed via the Debian package and have the latest stable version 2.19.10.10370.

I modified `/usr/bin/mattercontrol` to add the `--debug` flag to mono, and ran `mattercontrol` via the terminal, but it produced zero extra logs, error or otherwise. I did notice an error about a missing library, which went away when I installed the package with `sudo apt-get install libcanberra-gtk-module`, but this did _not_ fix the problem.

It is able to write to files in `~/.local/share/Mattercontrol`, and thanks to @gvcastellano's suggestion I was able to find what I needed after a lot of trial and error and before/after use of `find`.

FWIW @inferont, I'm able to open local .stl files.

I don't think it matters but I have the Pulse XE.