MatterHackers/MatterControl #4774: Top solid layers

m9mitch, 2020-06-16:
I usually keep my Top and Bottom Solid Layers set at 1.2mm and layer thickness at 0.2, Many times the slicer will only make 2 Top Solid Layers. I tried setting the Top Solid Layers to 6 and it still  only makes 2 but does make 6 on the bottom. This happens whether or not there are enough layers to require infill so sometimes there is infill even if there are 12 or less total layers where I would expect all solid layers ( 6 bottom and 6 top for 12 total layers). I would be pleased even if there was a work around for this.
I do have better success with your slicer than I did with Cura and I am able to do a lot of design and modifications with the simple tools you provide (and the price is right).
Thank you

tupc10, 2020-06-17:
Hello, first of all thanks for the great work of MatterHackers team! As @m9mitch said, I also prefer to use MatterSlice instead of Cura, Slicer and other existing slicers tools, it is a great tool and gives me excelent results! But, since the last update of June, I started to have this top layers bug too.. In my case it was even strange, because the slicer generated infill between top layers, and since then I was forced me to rollback to the previous release.. Please, take a look at this as soon as possible as I want to be up to date with all your releases! Thanks beforehand and best regards!

larsbrubaker, 2020-06-17:
If either (or both) of you could possibly export your profile and the stl that is causing this error we will be happy to get it fixed. You will have to .zip them and add them to this issue.

Also, if you have picture of the printed part screenshot of the gcode you can put that in the response as well. The more information the better.

Thank you.

m9mitch, 2020-06-18:
I’m sorry I am not more help, I have not been able to figure out how to export my profile.

 

Mitch

larsbrubaker, 2020-06-18:
Thanks for trying. Here is how you export your printer settings.

[image]

m9mitch, 2020-06-18:
Lars,

Here are the files, I will try to send the gcode in a separate email if your server doesn’t reject it for being too big. Let me know if you could use anything else.

 

Thanks,

m9mitch

m9mitch, 2020-06-18:
Gcode attached

 

m9mitch

larsbrubaker, 2020-06-18:
It seems GitHub will not let you add files as attachments. You will need to open in GitHub and add any files. Thank you.

m9mitch, 2020-06-18:
[m9mitch.zip](https://github.com/MatterHackers/MatterControl/files/4800929/m9mitch.zip)
[m9mitch gcode.zip](https://github.com/MatterHackers/MatterControl/files/4800930/m9mitch.gcode.zip)

larsbrubaker, 2020-06-19:
Thank you for the report. We will get a new test created for this and get it fixed ASAP.

larsbrubaker, 2020-06-20:
This has now been fixed. You can currently find the fix in the Alpha channel. Please let me know if this resolves the issue.

Thanks again.

[image]

m9mitch, 2020-06-20:
Great! Thank you

 

m9mitch

 

 <https://user-images.githubusercontent.com/1158332/85187072-b4954d00-b251-11ea-955b-6e52db58bf96.gif>

tupc10, 2020-06-21:
Hello, sorry for answering so late.. I tested the alpha version too and the problem seems to be resolved, great job @larsbrubaker !! Thank you very much for solving the problem so quickly! Best regards!

closed 2020-06-21 by larsbrubaker