MatterHackers/MatterControl #4777: Random start/end option for perimeters

tupc10, 2020-06-21:
Feature Request
===============

**Problem**
Hello, I am requesting this because I am getting very ugly Z scars on my 3D prints after MatterControl's June update that did not happen on the previous version. I realize that the Z scars could probably be reduced by improving the calibration of my 3D printers, but since it was done by default on the previous version I dont think it will be hard to implement, and maybe this feature could help other users facing similar printing results.

**Solution**
I would like an option for MatterSlice to start and end the perimeters at random points, and not only vertically aligned as it is being done by June's update of MatterControl.

3ddruckermedartis, 2020-09-03:
I would also need this one

3ddruckermedartis, 2020-09-08:
#

3ddruckermedartis, 2020-11-23:
Up