MatterHackers/MatterControl #4790: Linux X11 Error (BadMatch)

ponceto, 2020-07-01:
Hi,

I've installed MatterControl 2.0 on a fresh installed Linux Mint 20. The application doesn't start and complain about an X11 error. 

I'm using Nvidia drivers and the screen is using the default visual (TrueColor, 24-bits).

Here is the stack trace :

```
X11 Error encountered: 
  Error: BadMatch (invalid parameter attributes)
  Request:     151 (5)
  Resource ID: 0x4600014
  Serial:      262
  Hwnd:        Hwnd, Mapped:False ClientWindow:0x4600014, WholeWindow:0x4600013, Zombie=False, Parent:[Hwnd, Mapped:False ClientWindow:0x4600012, WholeWindow:0x4600011, Zombie=False, Parent:[<null>]]
  Control:     0  at System.Environment.get_StackTrace () [0x00000] in <12b418a7818c4ca0893feeaaf67f1e7f>:0 
  at System.Windows.Forms.XplatUIX11.HandleError (System.IntPtr display, System.Windows.Forms.XErrorEvent& error_event) [0x00000] in <a3daa9b84fd241a497578a25f68bc3c7>:0 
  at OpenTK.Platform.X11.Glx.MakeCurrent (System.IntPtr , System.IntPtr , System.IntPtr ) [0x00000] in <d85857bc66524258b04ff0bdaeed67db>:0 
  at OpenTK.Platform.X11.Glx.MakeCurrent (System.IntPtr display, System.IntPtr drawable, OpenTK.ContextHandle context) [0x00000] in <d85857bc66524258b04ff0bdaeed67db>:0 
  at OpenTK.Platform.X11.X11GLContext.MakeCurrent (OpenTK.Platform.IWindowInfo window) [0x00000] in <d85857bc66524258b04ff0bdaeed67db>:0 
  at OpenTK.Graphics.GraphicsContext.MakeCurrent (OpenTK.Platform.IWindowInfo window) [0x00000] in <d85857bc66524258b04ff0bdaeed67db>:0 
  at OpenTK.GLControl.MakeCurrent () [0x00000] in <33e0a290e5404e8e8877ff9c93b1ec2f>:0 
  at OpenTK.GLControl.OnHandleCreated (System.EventArgs e) [0x00000] in <33e0a290e5404e8e8877ff9c93b1ec2f>:0 
  at System.Windows.Forms.Control.WmCreate (System.Windows.Forms.Message& m) [0x00000] in <a3daa9b84fd241a497578a25f68bc3c7>:0 
  at System.Windows.Forms.Control.WndProc (System.Windows.Forms.Message& m) [0x00000] in <a3daa9b84fd241a497578a25f68bc3c7>:0 
  at System.Windows.Forms.ScrollableControl.WndProc (System.Windows.Forms.Message& m) [0x00000] in <a3daa9b84fd241a497578a25f68bc3c7>:0 
  at System.Windows.Forms.ContainerControl.WndProc (System.Windows.Forms.Message& m) [0x00000] in <a3daa9b84fd241a497578a25f68bc3c7>:0 
  at System.Windows.Forms.UserControl.WndProc (System.Windows.Forms.Message& m) [0x00000] in <a3daa9b84fd241a497578a25f68bc3c7>:0 
  at System.Windows.Forms.Control+ControlWindowTarget.OnMessage (System.Windows.Forms.Message& m) [0x00000] in <a3daa9b84fd241a497578a25f68bc3c7>:0 
  at System.Windows.Forms.Control+ControlNativeWindow.WndProc (System.Windows.Forms.Message& m) [0x00000] in <a3daa9b84fd241a497578a25f68bc3c7>:0 
  at System.Windows.Forms.NativeWindow.WndProc (System.IntPtr hWnd, System.Windows.Forms.Msg msg, System.IntPtr wParam, System.IntPtr lParam) [0x00000] in <a3daa9b84fd241a497578a25f68bc3c7>:0 
  at System.Windows.Forms.XplatUIX11.SendMessage (System.IntPtr hwnd, System.Windows.Forms.Msg message, System.IntPtr wParam, System.IntPtr lParam) [0x00000] in <a3daa9b84fd241a497578a25f68bc3c7>:0 
  at System.Windows.Forms.XplatUIX11.CreateWindow (System.Windows.Forms.CreateParams cp) [0x00000] in <a3daa9b84fd241a497578a25f68bc3c7>:0 
  at System.Windows.Forms.XplatUI.CreateWindow (System.Windows.Forms.CreateParams cp) [0x00000] in <a3daa9b84fd241a497578a25f68bc3c7>:0 
  at System.Windows.Forms.NativeWindow.CreateHandle (System.Windows.Forms.CreateParams cp) [0x00000] in <a3daa9b84fd241a497578a25f68bc3c7>:0 
  at System.Windows.Forms.Control.CreateHandle () [0x00000] in <a3daa9b84fd241a497578a25f68bc3c7>:0 
  at System.Windows.Forms.Control.CreateControl () [0x00000] in <a3daa9b84fd241a497578a25f68bc3c7>:0 
  at System.Windows.Forms.Control.CreateControl () [0x00000] in <a3daa9b84fd241a497578a25f68bc3c7>:0 
  at System.Windows.Forms.Control.SetVisibleCore (System.Boolean value) [0x00000] in <a3daa9b84fd241a497578a25f68bc3c7>:0 
  at System.Windows.Forms.Form.SetVisibleCore (System.Boolean value) [0x00000] in <a3daa9b84fd241a497578a25f68bc3c7>:0 
  at MatterHackers.Agg.UI.WinformsSystemWindow.SetVisibleCore (System.Boolean value) [0x00000] in <712ffdb4c0144e8c88c0ce33a0b20977>:0 
  at System.Windows.Forms.Control.set_Visible (System.Boolean value) [0x00000] in <a3daa9b84fd241a497578a25f68bc3c7>:0 
  at System.Windows.Forms.Control.Show () [0x00000] in <a3daa9b84fd241a497578a25f68bc3c7>:0 
  at MatterHackers.Agg.UI.WinformsSystemWindow.Show () [0x00000] in <712ffdb4c0144e8c88c0ce33a0b20977>:0 
  at MatterHackers.Agg.UI.WinformsSystemWindow.ShowSystemWindow (MatterHackers.Agg.UI.SystemWindow systemWindow) [0x00000] in <712ffdb4c0144e8c88c0ce33a0b20977>:0 
  at MatterHackers.Agg.UI.SingleWindowProvider.ShowSystemWindow (MatterHackers.Agg.UI.SystemWindow systemWindow) [0x00000] in <49bb6fd6740f4762a381356153fe2e02>:0 
  at MatterHackers.MatterControl.WinformsSingleWindowProvider.ShowSystemWindow (MatterHackers.Agg.UI.SystemWindow systemWindow) [0x00000] in <16b4831ddba24ef8b2ca5ca9a43af3cd>:0 
  at MatterHackers.Agg.UI.SystemWindow.ShowAsSystemWindow () [0x00000] in <a5c2db3666d84398b1090e2f2dfee474>:0 
  at MatterHackers.MatterControl.Program.Main (System.String[] args) [0x00000] in <e58f04b6857344df9781155f8ee2c0c8>:0 

Time to '': 800
X11 Error encountered: 
  Error: BadMatch (invalid parameter attributes)
  Request:     151 (5)
  Resource ID: 0x4600014
  Serial:      353
  Hwnd:        Hwnd, Mapped:True ClientWindow:0x4600014, WholeWindow:0x4600013, Zombie=False, Parent:[Hwnd, Mapped:True ClientWindow:0x4600012, WholeWindow:0x4600011, Zombie=False, Parent:[<null>]]
  Control:     0  at System.Environment.get_StackTrace () [0x00000] in <12b418a7818c4ca0893feeaaf67f1e7f>:0 
  at System.Windows.Forms.XplatUIX11.HandleError (System.IntPtr display, System.Windows.Forms.XErrorEvent& error_event) [0x00000] in <a3daa9b84fd241a497578a25f68bc3c7>:0 
  at OpenTK.Platform.X11.Glx.MakeCurrent (System.IntPtr , System.IntPtr , System.IntPtr ) [0x00000] in <d85857bc66524258b04ff0bdaeed67db>:0 
  at OpenTK.Platform.X11.Glx.MakeCurrent (System.IntPtr display, System.IntPtr drawable, OpenTK.ContextHandle context) [0x00000] in <d85857bc66524258b04ff0bdaeed67db>:0 
  at OpenTK.Platform.X11.X11GLContext.MakeCurrent (OpenTK.Platform.IWindowInfo window) [0x00000] in <d85857bc66524258b04ff0bdaeed67db>:0 
  at OpenTK.Graphics.GraphicsContext.MakeCurrent (OpenTK.Platform.IWindowInfo window) [0x00000] in <d85857bc66524258b04ff0bdaeed67db>:0 
  at OpenTK.GLControl.MakeCurrent () [0x00000] in <33e0a290e5404e8e8877ff9c93b1ec2f>:0 
  at MatterHackers.Agg.UI.AggGLControl.MakeCurrent () [0x00000] in <712ffdb4c0144e8c88c0ce33a0b20977>:0 
  at MatterHackers.Agg.UI.OpenGLSystemWindow.OnPaint (System.Windows.Forms.PaintEventArgs paintEventArgs) [0x00000] in <712ffdb4c0144e8c88c0ce33a0b20977>:0 
  at System.Windows.Forms.Control.WmPaint (System.Windows.Forms.Message& m) [0x00000] in <a3daa9b84fd241a497578a25f68bc3c7>:0 
  at System.Windows.Forms.Control.WndProc (System.Windows.Forms.Message& m) [0x00000] in <a3daa9b84fd241a497578a25f68bc3c7>:0 
  at System.Windows.Forms.ScrollableControl.WndProc (System.Windows.Forms.Message& m) [0x00000] in <a3daa9b84fd241a497578a25f68bc3c7>:0 
  at System.Windows.Forms.ContainerControl.WndProc (System.Windows.Forms.Message& m) [0x00000] in <a3daa9b84fd241a497578a25f68bc3c7>:0 
[... 45 lines cut ...]
  at MatterHackers.Agg.UI.GuiWidget.<CloseOnIdle>b__391_0 () [0x00000] in <a5c2db3666d84398b1090e2f2dfee474>:0 
  at MatterHackers.Agg.UI.UiThread.InvokePendingActions () [0x00000] in <a5c2db3666d84398b1090e2f2dfee474>:0 
  at System.Reflection.RuntimeMethodInfo.InternalInvoke (System.Reflection.RuntimeMethodInfo , System.Object , System.Object[] , System.Exception& ) [0x00000] in <12b418a7818c4ca0893feeaaf67f1e7f>:0 
  at System.Reflection.RuntimeMethodInfo.Invoke (System.Object obj, System.Reflection.BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) [0x00000] in <12b418a7818c4ca0893feeaaf67f1e7f>:0 
  at System.Reflection.MethodBase.Invoke (System.Object obj, System.Object[] parameters) [0x00000] in <12b418a7818c4ca0893feeaaf67f1e7f>:0 
  at System.Delegate.DynamicInvokeImpl (System.Object[] args) [0x00000] in <12b418a7818c4ca0893feeaaf67f1e7f>:0 
  at System.MulticastDelegate.DynamicInvokeImpl (System.Object[] args) [0x00000] in <12b418a7818c4ca0893feeaaf67f1e7f>:0 
  at System.Delegate.DynamicInvoke (System.Object[] args) [0x00000] in <12b418a7818c4ca0893feeaaf67f1e7f>:0 
[... cut ...]
  at MatterHackers.Agg.UI.SingleWindowProvider.ShowSystemWindow (MatterHackers.Agg.UI.SystemWindow systemWindow) [0x00000] in <49bb6fd6740f4762a381356153fe2e02>:0 
  at MatterHackers.MatterControl.WinformsSingleWindowProvider.ShowSystemWindow (MatterHackers.Agg.UI.SystemWindow systemWindow) [0x00000] in <16b4831ddba24ef8b2ca5ca9a43af3cd>:0 
  at MatterHackers.Agg.UI.SystemWindow.ShowAsSystemWindow () [0x00000] in <a5c2db3666d84398b1090e2f2dfee474>:0 
  at MatterHackers.MatterControl.Program.Main (System.String[] args) [0x00000] in <e58f04b6857344df9781155f8ee2c0c8>:0 
```

didli, 2020-07-18:
Yep. Still there : https://github.com/MatterHackers/MatterControl/issues/4682
Building from source result in the same issue :
`X11 Error encountered: 
  Error: BadMatch (invalid parameter attributes)
  Request:     150 (5)
  Resource ID: 0x7200011
  Serial:      252
  Hwnd:        Hwnd, Mapped:False ClientWindow:0x7200011, WholeWindow:0x7200010, Zombie=False, Parent:[Hwnd, Mapped:False ClientWindow:0x720000F, WholeWindow:0x720000E, Zombie=False, Parent:[<null>]]
  Control:     0  at System.Environment.get_StackTrace () [0x00000] in <e50d21fbfba24408b2ca69b0d707999a>:0 
  at System.Windows.Forms.XplatUIX11.HandleError (System.IntPtr display, System.Windows.Forms.XErrorEvent& error_event) [0x00000] in <91835e866d3040be91350e07c4485584>:0 
  at OpenTK.Platform.X11.Glx.MakeCurrent (System.IntPtr , System.IntPtr , System.IntPtr ) [0x00000] in <d85857bc66524258b04ff0bdaeed67db>:0 
  at OpenTK.Platform.X11.Glx.MakeCurrent (System.IntPtr display, System.IntPtr drawable, OpenTK.ContextHandle context) [0x00000] in <d85857bc66524258b04ff0bdaeed67db>:0 
  at OpenTK.Platform.X11.X11GLContext.MakeCurrent (OpenTK.Platform.IWindowInfo window) [0x00000] in <d85857bc66524258b04ff0bdaeed67db>:0 
  at OpenTK.Graphics.GraphicsContext.MakeCurrent (OpenTK.Platform.IWindowInfo window) [0x00000] in <d85857bc66524258b04ff0bdaeed67db>:0 
  at OpenTK.GLControl.MakeCurrent () [0x00000] in <33e0a290e5404e8e8877ff9c93b1ec2f>:0 
  at OpenTK.GLControl.OnHandleCreated (System.EventArgs e) [0x00000] in <33e0a290e5404e8e8877ff9c93b1ec2f>:0 
  at System.Windows.Forms.Control.WmCreate (System.Windows.Forms.Message& m) [0x00000] in <91835e866d3040be91350e07c4485584>:0 
  at System.Windows.Forms.Control.WndProc (System.Windows.Forms.Message& m) [0x00000] in <91835e866d3040be91350e07c4485584>:0 
  at System.Windows.Forms.ScrollableControl.WndProc (System.Windows.Forms.Message& m) [0x00000] in <91835e866d3040be91350e07c4485584>:0 
  at System.Windows.Forms.ContainerControl.WndProc (System.Windows.Forms.Message& m) [0x00000] in <91835e866d3040be91350e07c4485584>:0 
  at System.Windows.Forms.UserControl.WndProc (System.Windows.Forms.Message& m) [0x00000] in <91835e866d3040be91350e07c4485584>:0 
  at System.Windows.Forms.Control+ControlWindowTarget.OnMessage (System.Windows.Forms.Message& m) [0x00000] in <91835e866d3040be91350e07c4485584>:0 
  at System.Windows.Forms.Control+ControlNativeWindow.WndProc (System.Windows.Forms.Message& m) [0x00000] in <91835e866d3040be91350e07c4485584>:0 
  at System.Windows.Forms.NativeWindow.WndProc (System.IntPtr hWnd, System.Windows.Forms.Msg msg, System.IntPtr wParam, System.IntPtr lParam) [0x00000] in <91835e866d3040be91350e07c4485584>:0 
  at System.Windows.Forms.XplatUIX11.SendMessage (System.IntPtr hwnd, System.Windows.Forms.Msg message, System.IntPtr wParam, System.IntPtr lParam) [0x00000] in <91835e866d3040be91350e07c4485584>:0 
  at System.Windows.Forms.XplatUIX11.CreateWindow (System.Windows.Forms.CreateParams cp) [0x00000] in <91835e866d3040be91350e07c4485584>:0 
  at System.Windows.Forms.XplatUI.CreateWindow (System.Windows.Forms.CreateParams cp) [0x00000] in <91835e866d3040be91350e07c4485584>:0 
  at System.Windows.Forms.NativeWindow.CreateHandle (System.Windows.Forms.CreateParams cp) [0x00000] in <91835e866d3040be91350e07c4485584>:0 
  at System.Windows.Forms.Control.CreateHandle () [0x00000] in <91835e866d3040be91350e07c4485584>:0 
  at System.Windows.Forms.Control.CreateControl () [0x00000] in <91835e866d3040be91350e07c4485584>:0 
  at System.Windows.Forms.Control.CreateControl () [0x00000] in <91835e866d3040be91350e07c4485584>:0 
  at System.Windows.Forms.Control.SetVisibleCore (System.Boolean value) [0x00000] in <91835e866d3040be91350e07c4485584>:0 
  at System.Windows.Forms.Form.SetVisibleCore (System.Boolean value) [0x00000] in <91835e866d3040be91350e07c4485584>:0 
  at MatterHackers.Agg.UI.WinformsSystemWindow.SetVisibleCore (System.Boolean value) [0x00000] in <3880858f46bf4e7585d7990a40fb346e>:0 
  at System.Windows.Forms.Control.set_Visible (System.Boolean value) [0x00000] in <91835e866d3040be91350e07c4485584>:0 
  at System.Windows.Forms.Control.Show () [0x00000] in <91835e866d3040be91350e07c4485584>:0 
  at MatterHackers.Agg.UI.WinformsSystemWindow.Show () [0x00000] in <3880858f46bf4e7585d7990a40fb346e>:0 
  at MatterHackers.Agg.UI.WinformsSystemWindow.ShowSystemWindow (MatterHackers.Agg.UI.SystemWindow systemWindow) [0x00000] in <3880858f46bf4e7585d7990a40fb346e>:0 
  at MatterHackers.Agg.UI.SingleWindowProvider.ShowSystemWindow (MatterHackers.Agg.UI.SystemWindow systemWindow) [0x00000] in <cd3036edef1642a2bc45a791a781801a>:0 
  at MatterHackers.MatterControl.WinformsSingleWindowProvider.ShowSystemWindow (MatterHackers.Agg.UI.SystemWindow systemWindow) [0x00000] in <089d2105cf584de3ab47b25c342c770d>:0 
  at MatterHackers.Agg.UI.SystemWindow.ShowAsSystemWindow () [0x00000] in <2c7a0206278f47928849fb87c13cbbc5>:0 
  at MatterHackers.MatterControl.Program.Main (System.String[] args) [0x00000] in <104106100d35454e8f9107e8568e66f6>:0 

OpenTK.Graphics.GraphicsContextException: Failed to make context current.
  at OpenTK.Platform.X11.X11GLContext.MakeCurrent (OpenTK.Platform.IWindowInfo window) [0x0010e] in <d85857bc66524258b04ff0bdaeed67db>:0 
  at OpenTK.Graphics.GraphicsContext.MakeCurrent (OpenTK.Platform.IWindowInfo window) [0x00000] in <d85857bc66524258b04ff0bdaeed67db>:0 
  at OpenTK.GLControl.MakeCurrent () [0x00017] in <33e0a290e5404e8e8877ff9c93b1ec2f>:0 
[... cut ...]
  at System.Windows.Forms.Control+ControlWindowTarget.OnMessage (System.Windows.Forms.Message& m) [0x00000] in <91835e866d3040be91350e07c4485584>:0 
  at System.Windows.Forms.Control+ControlNativeWindow.WndProc (System.Windows.Forms.Message& m) [0x0000b] in <91835e866d3040be91350e07c4485584>:0 
  at System.Windows.Forms.NativeWindow.WndProc (System.IntPtr hWnd, System.Windows.Forms.Msg msg, System.IntPtr wParam, System.IntPtr lParam) [0x00085] in <91835e866d3040be91350e07c4485584>:0 
Time to '': 190
Thread was being aborted.`

digitables, 2020-10-05:
I'm also getting a similar issue.

X11 Error encountered: 
  Error: BadMatch (invalid parameter attributes)
  Request:     152 (5)
  Resource ID: 0x2A00011
  Serial:      255
  Hwnd:        Hwnd, Mapped:False ClientWindow:0x2A00011, WholeWindow:0x2A00010, Zombie=False, Parent:[Hwnd, Mapped:False ClientWindow:0x2A0000F, WholeWindow:0x2A0000E, Zombie=False, Parent:[<null>]]
  Control:     0  at System.Environment.get_StackTrace () [0x00000] in <671ef0784b04474c83bb500a849492f2>:0 
  at System.Windows.Forms.XplatUIX11.HandleError (System.IntPtr display, System.Windows.Forms.XErrorEvent& error_event) [0x00000] in <313abb6e08e943459ea3b455bc2f9adb>:0 
  at OpenTK.Platform.X11.Glx.MakeCurrent (System.IntPtr , System.IntPtr , System.IntPtr ) [0x00000] in <d85857bc66524258b04ff0bdaeed67db>:0 
  at OpenTK.Platform.X11.Glx.MakeCurrent (System.IntPtr display, System.IntPtr drawable, OpenTK.ContextHandle context) [0x00000] in <d85857bc66524258b04ff0bdaeed67db>:0 
  at OpenTK.Platform.X11.X11GLContext.MakeCurrent (OpenTK.Platform.IWindowInfo window) [0x00000] in <d85857bc66524258b04ff0bdaeed67db>:0 
  at OpenTK.Graphics.GraphicsContext.MakeCurrent (OpenTK.Platform.IWindowInfo window) [0x00000] in <d85857bc66524258b04ff0bdaeed67db>:0 
  at OpenTK.GLControl.MakeCurrent () [0x00000] in <33e0a290e5404e8e8877ff9c93b1ec2f>:0 
  at OpenTK.GLControl.OnHandleCreated (System.EventArgs e) [0x00000] in <33e0a290e5404e8e8877ff9c93b1ec2f>:0 
  at System.Windows.Forms.Control.WmCreate (System.Windows.Forms.Message& m) [0x00000] in <313abb6e08e943459ea3b455bc2f9adb>:0 
  at System.Windows.Forms.Control.WndProc (System.Windows.Forms.Message& m) [0x00000] in <313abb6e08e943459ea3b455bc2f9adb>:0 
  at System.Windows.Forms.ScrollableControl.WndProc (System.Windows.Forms.Message& m) [0x00000] in <313abb6e08e943459ea3b455bc2f9adb>:0 
  at System.Windows.Forms.ContainerControl.WndProc (System.Windows.Forms.Message& m) [0x00000] in <313abb6e08e943459ea3b455bc2f9adb>:0 
  at System.Windows.Forms.UserControl.WndProc (System.Windows.Forms.Message& m) [0x00000] in <313abb6e08e943459ea3b455bc2f9adb>:0 
  at System.Windows.Forms.Control+ControlWindowTarget.OnMessage (System.Windows.Forms.Message& m) [0x00000] in <313abb6e08e943459ea3b455bc2f9adb>:0 
  at System.Windows.Forms.Control+ControlNativeWindow.WndProc (System.Windows.Forms.Message& m) [0x00000] in <313abb6e08e943459ea3b455bc2f9adb>:0 
  at System.Windows.Forms.NativeWindow.WndProc (System.IntPtr hWnd, System.Windows.Forms.Msg msg, System.IntPtr wParam, System.IntPtr lParam) [0x00000] in <313abb6e08e943459ea3b455bc2f9adb>:0 
  at System.Windows.Forms.XplatUIX11.SendMessage (System.IntPtr hwnd, System.Windows.Forms.Msg message, System.IntPtr wParam, System.IntPtr lParam) [0x00000] in <313abb6e08e943459ea3b455bc2f9adb>:0 
  at System.Windows.Forms.XplatUIX11.CreateWindow (System.Windows.Forms.CreateParams cp) [0x00000] in <313abb6e08e943459ea3b455bc2f9adb>:0 
  at System.Windows.Forms.XplatUI.CreateWindow (System.Windows.Forms.CreateParams cp) [0x00000] in <313abb6e08e943459ea3b455bc2f9adb>:0 
  at System.Windows.Forms.NativeWindow.CreateHandle (System.Windows.Forms.CreateParams cp) [0x00000] in <313abb6e08e943459ea3b455bc2f9adb>:0 
  at System.Windows.Forms.Control.CreateHandle () [0x00000] in <313abb6e08e943459ea3b455bc2f9adb>:0 
  at System.Windows.Forms.Control.CreateControl () [0x00000] in <313abb6e08e943459ea3b455bc2f9adb>:0 
  at System.Windows.Forms.Control.CreateControl () [0x00000] in <313abb6e08e943459ea3b455bc2f9adb>:0 
  at System.Windows.Forms.Control.SetVisibleCore (System.Boolean value) [0x00000] in <313abb6e08e943459ea3b455bc2f9adb>:0 
  at System.Windows.Forms.Form.SetVisibleCore (System.Boolean value) [0x00000] in <313abb6e08e943459ea3b455bc2f9adb>:0 
  at MatterHackers.Agg.UI.WinformsSystemWindow.SetVisibleCore (System.Boolean value) [0x00000] in <712ffdb4c0144e8c88c0ce33a0b20977>:0 
  at System.Windows.Forms.Control.set_Visible (System.Boolean value) [0x00000] in <313abb6e08e943459ea3b455bc2f9adb>:0 
  at System.Windows.Forms.Control.Show () [0x00000] in <313abb6e08e943459ea3b455bc2f9adb>:0 
  at MatterHackers.Agg.UI.WinformsSystemWindow.Show () [0x00000] in <712ffdb4c0144e8c88c0ce33a0b20977>:0 
  at MatterHackers.Agg.UI.WinformsSystemWindow.ShowSystemWindow (MatterHackers.Agg.UI.SystemWindow systemWindow) [0x00000] in <712ffdb4c0144e8c88c0ce33a0b20977>:0 
  at MatterHackers.Agg.UI.SingleWindowProvider.ShowSystemWindow (MatterHackers.Agg.UI.SystemWindow systemWindow) [0x00000] in <49bb6fd6740f4762a381356153fe2e02>:0 
  at MatterHackers.MatterControl.WinformsSingleWindowProvider.ShowSystemWindow (MatterHackers.Agg.UI.SystemWindow systemWindow) [0x00000] in <16b4831ddba24ef8b2ca5ca9a43af3cd>:0 
  at MatterHackers.Agg.UI.SystemWindow.ShowAsSystemWindow () [0x00000] in <a5c2db3666d84398b1090e2f2dfee474>:0 
  at MatterHackers.MatterControl.Program.Main (System.String[] args) [0x00000] in <e58f04b6857344df9781155f8ee2c0c8>:0 

Time to '': 898
X11 Error encountered: 
  Error: BadMatch (invalid parameter attributes)
  Request:     152 (5)
  Resource ID: 0x2A00011
  Serial:      311
  Hwnd:        Hwnd, Mapped:True ClientWindow:0x2A00011, WholeWindow:0x2A00010, Zombie=False, Parent:[Hwnd, Mapped:True ClientWindow:0x2A0000F, WholeWindow:0x2A0000E, Zombie=False, Parent:[<null>]]
  Control:     0  at System.Environment.get_StackTrace () [0x00000] in <671ef0784b04474c83bb500a849492f2>:0 
[... cut ...]
  at MatterHackers.Agg.UI.SingleWindowProvider.ShowSystemWindow (MatterHackers.Agg.UI.SystemWindow systemWindow) [0x00000] in <49bb6fd6740f4762a381356153fe2e02>:0 
  at MatterHackers.MatterControl.WinformsSingleWindowProvider.ShowSystemWindow (MatterHackers.Agg.UI.SystemWindow systemWindow) [0x00000] in <16b4831ddba24ef8b2ca5ca9a43af3cd>:0 
  at MatterHackers.Agg.UI.SystemWindow.ShowAsSystemWindow () [0x00000] in <a5c2db3666d84398b1090e2f2dfee474>:0 
  at MatterHackers.MatterControl.Program.Main (System.String[] args) [0x00000] in <e58f04b6857344df9781155f8ee2c0c8>:0

Shadow3641, 2020-10-23:
Here is mine.
I am also using NVIDIA drivers 450

X11 Error encountered: 
  Error: BadMatch (invalid parameter attributes)
  Request:     151 (5)
  Resource ID: 0x3600014
  Serial:      270
  Hwnd:        Hwnd, Mapped:False ClientWindow:0x3600014, WholeWindow:0x3600013, Zombie=False, Parent:[Hwnd, Mapped:False ClientWindow:0x3600012, WholeWindow:0x3600011, Zombie=False, Parent:[<null>]]
  Control:     0  at System.Environment.get_StackTrace () [0x00000] in <9f0df102fe6e4cfea29d2e46f585d8a5>:0 
  at System.Windows.Forms.XplatUIX11.HandleError (System.IntPtr display, System.Windows.Forms.XErrorEvent& error_event) [0x00000] in <47c199caaa3c4c6c9c48a7c6ebb3fd29>:0 
  at OpenTK.Platform.X11.Glx.MakeCurrent (System.IntPtr , System.IntPtr , System.IntPtr ) [0x00000] in <d85857bc66524258b04ff0bdaeed67db>:0 
  at OpenTK.Platform.X11.Glx.MakeCurrent (System.IntPtr display, System.IntPtr drawable, OpenTK.ContextHandle context) [0x00000] in <d85857bc66524258b04ff0bdaeed67db>:0 
  at OpenTK.Platform.X11.X11GLContext.MakeCurrent (OpenTK.Platform.IWindowInfo window) [0x00000] in <d85857bc66524258b04ff0bdaeed67db>:0 
  at OpenTK.Graphics.GraphicsContext.MakeCurrent (OpenTK.Platform.IWindowInfo window) [0x00000] in <d85857bc66524258b04ff0bdaeed67db>:0 
  at OpenTK.GLControl.MakeCurrent () [0x00000] in <33e0a290e5404e8e8877ff9c93b1ec2f>:0 
  at OpenTK.GLControl.OnHandleCreated (System.EventArgs e) [0x00000] in <33e0a290e5404e8e8877ff9c93b1ec2f>:0 
  at System.Windows.Forms.Control.WmCreate (System.Windows.Forms.Message& m) [0x00000] in <47c199caaa3c4c6c9c48a7c6ebb3fd29>:0 
  at System.Windows.Forms.Control.WndProc (System.Windows.Forms.Message& m) [0x00000] in <47c199caaa3c4c6c9c48a7c6ebb3fd29>:0 
  at System.Windows.Forms.ScrollableControl.WndProc (System.Windows.Forms.Message& m) [0x00000] in <47c199caaa3c4c6c9c48a7c6ebb3fd29>:0 
  at System.Windows.Forms.ContainerControl.WndProc (System.Windows.Forms.Message& m) [0x00000] in <47c199caaa3c4c6c9c48a7c6ebb3fd29>:0 
  at System.Windows.Forms.UserControl.WndProc (System.Windows.Forms.Message& m) [0x00000] in <47c199caaa3c4c6c9c48a7c6ebb3fd29>:0 
  at System.Windows.Forms.Control+ControlWindowTarget.OnMessage (System.Windows.Forms.Message& m) [0x00000] in <47c199caaa3c4c6c9c48a7c6ebb3fd29>:0 
  at System.Windows.Forms.Control+ControlNativeWindow.WndProc (System.Windows.Forms.Message& m) [0x00000] in <47c199caaa3c4c6c9c48a7c6ebb3fd29>:0 
  at System.Windows.Forms.NativeWindow.WndProc (System.IntPtr hWnd, System.Windows.Forms.Msg msg, System.IntPtr wParam, System.IntPtr lParam) [0x00000] in <47c199caaa3c4c6c9c48a7c6ebb3fd29>:0 
  at System.Windows.Forms.XplatUIX11.SendMessage (System.IntPtr hwnd, System.Windows.Forms.Msg message, System.IntPtr wParam, System.IntPtr lParam) [0x00000] in <47c199caaa3c4c6c9c48a7c6ebb3fd29>:0 
  at System.Windows.Forms.XplatUIX11.CreateWindow (System.Windows.Forms.CreateParams cp) [0x00000] in <47c199caaa3c4c6c9c48a7c6ebb3fd29>:0 
  at System.Windows.Forms.XplatUI.CreateWindow (System.Windows.Forms.CreateParams cp) [0x00000] in <47c199caaa3c4c6c9c48a7c6ebb3fd29>:0 
  at System.Windows.Forms.NativeWindow.CreateHandle (System.Windows.Forms.CreateParams cp) [0x00000] in <47c199caaa3c4c6c9c48a7c6ebb3fd29>:0 
  at System.Windows.Forms.Control.CreateHandle () [0x00000] in <47c199caaa3c4c6c9c48a7c6ebb3fd29>:0 
  at System.Windows.Forms.Control.CreateControl () [0x00000] in <47c199caaa3c4c6c9c48a7c6ebb3fd29>:0 
  at System.Windows.Forms.Control.CreateControl () [0x00000] in <47c199caaa3c4c6c9c48a7c6ebb3fd29>:0 
  at System.Windows.Forms.Control.SetVisibleCore (System.Boolean value) [0x00000] in <47c199caaa3c4c6c9c48a7c6ebb3fd29>:0 
  at System.Windows.Forms.Form.SetVisibleCore (System.Boolean value) [0x00000] in <47c199caaa3c4c6c9c48a7c6ebb3fd29>:0 
  at MatterHackers.Agg.UI.WinformsSystemWindow.SetVisibleCore (System.Boolean value) [0x00000] in <712ffdb4c0144e8c88c0ce33a0b20977>:0 
  at System.Windows.Forms.Control.set_Visible (System.Boolean value) [0x00000] in <47c199caaa3c4c6c9c48a7c6ebb3fd29>:0 
  at System.Windows.Forms.Control.Show () [0x00000] in <47c199caaa3c4c6c9c48a7c6ebb3fd29>:0 
  at MatterHackers.Agg.UI.WinformsSystemWindow.Show () [0x00000] in <712ffdb4c0144e8c88c0ce33a0b20977>:0 
  at MatterHackers.Agg.UI.WinformsSystemWindow.ShowSystemWindow (MatterHackers.Agg.UI.SystemWindow systemWindow) [0x00000] in <712ffdb4c0144e8c88c0ce33a0b20977>:0 
  at MatterHackers.Agg.UI.SingleWindowProvider.ShowSystemWindow (MatterHackers.Agg.UI.SystemWindow systemWindow) [0x00000] in <49bb6fd6740f4762a381356153fe2e02>:0 
  at MatterHackers.MatterControl.WinformsSingleWindowProvider.ShowSystemWindow (MatterHackers.Agg.UI.SystemWindow systemWindow) [0x00000] in <16b4831ddba24ef8b2ca5ca9a43af3cd>:0 
  at MatterHackers.Agg.UI.SystemWindow.ShowAsSystemWindow () [0x00000] in <a5c2db3666d84398b1090e2f2dfee474>:0 
  at MatterHackers.MatterControl.Program.Main (System.String[] args) [0x00000] in <e58f04b6857344df9781155f8ee2c0c8>:0 

exception inside UnhandledException handler: (null) assembly:/usr/lib/mono/4.5/mscorlib.dll type:FileNotFoundException member:(null)

[ERROR] FATAL UNHANDLED EXCEPTION: System.IO.FileNotFoundException: Could not load file or assembly 'Microsoft.VisualBasic, Version=10.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a' or one of its dependencies.
File name: 'Microsoft.VisualBasic, Version=10.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a'
[... cut ...]
  at MatterHackers.Agg.UI.SingleWindowProvider.ShowSystemWindow (MatterHackers.Agg.UI.SystemWindow systemWindow) [0x00373] in <49bb6fd6740f4762a381356153fe2e02>:0 
  at MatterHackers.MatterControl.WinformsSingleWindowProvider.ShowSystemWindow (MatterHackers.Agg.UI.SystemWindow systemWindow) [0x00031] in <16b4831ddba24ef8b2ca5ca9a43af3cd>:0 
  at MatterHackers.Agg.UI.SystemWindow.ShowAsSystemWindow () [0x0002b] in <a5c2db3666d84398b1090e2f2dfee474>:0 
  at MatterHackers.MatterControl.Program.Main (System.String[] args) [0x0037f] in <e58f04b6857344df9781155f8ee2c0c8>:0

Marcus-Depew, 2020-10-27:
This bug is very pesky. Win10 version working on WINE, fairly stable.

CAHbKA-IV, 2021-02-20:
You can edit /usr/lib/mattercontrol/appsettings.json to override this bug. Application crashes while 'Color' and 'Depth' parameters has different values. I set this values to '24', and program works fine.