MatterHackers/MatterControl #4799: Printer without heated bed unable to start prints

jlewin, 2020-07-06:
**Repro**
1. Open MatterControl
2. Create or open a printer without a heated bed
3. Ensure Has Heated Bed is disabled
4. Start a test print

[image]

**Expected Behavior**
The printer should heat the hotend and start the print

**Actual Behavior**
The printer shows "Heating Bed..." and the print never starts
  [image]

jlewin, 2020-07-06:
Had to switch to a source build and put in guards around setting target temps to get my printer working. Not sure if you want to pick up the changes or if they're appropriate, but figured I'd share as my printer previously worked. Having stated that, the root of the problem is my config has a bed temp and with hasHeatedBed disabled, it's not displayed in the UI and unfortunately still has an effect.

This worked for me:
[patch.zip](https://github.com/MatterHackers/MatterControl/files/4876212/patch.zip)

larsbrubaker, 2020-07-11:
This patch has been commited

closed 2020-07-11 by larsbrubaker