MatterHackers/MatterControl #4805: Saved objects located in local library no longer load

ladouceurreflex, 2020-07-18:
Bug Report
============
**Steps to Reproduce**
1. Navigate to local library
2. Select a previously constructed object
3. Right click and select open in a new workspace.  

**Expected Behavior**
Object is loaded into the current workspace.

**Actual Behavior**
The workspace attempts to load the object as I see it in the object tree then it disappears without any feedback. 
The "Add to bed" option has the same effect.

This doesn't happen on all of my custom objects only some of my more recent objects.   I am able to export the objects to STL without an issue but I lose the capability to design the object.

**MatterControl Build Number**
2.20.6.10543

**Operating System Version**
Windows 10

**Printer Make/Model**
Pulse 3D

I don't know where local library files are created on my machine.   If you let me know where the local library objects are stored I can add the objects im trying to load.

Also, a path to the logs would be helpful.

larsbrubaker, 2020-07-18:
Can you save the .mcx file to your downloads folder and upload it to this issue?

MCX files reference .stl in your file system and do not make copies of them. It is possible that you have deleted or moved some of the files that it is looking for. If you upload it we will take a look to see what it is referencing.

ladouceurreflex, 2020-07-21:
Here is the MCX file I was able to locate after I used the move option on the object located in the Local Library folder.   Where are local library objects stored on the file system?

I haven't made any file system changes so I'm not sure how the references would have been broken.   I always only use MatterControl.

[3jfgm54t.zip](https://github.com/MatterHackers/MatterControl/files/4951191/3jfgm54t.zip)