MatterHackers/MatterControl #4808: GCode cache not invalidated after source STL changed

jlewin, 2020-07-29:
**Repo**
1. Open MatterControl
2. Drag a given file into MatterControl
3. Slice
4. Close MatterControl
5. Overwrite the original file with some visually identifiable change - (in my case using Fusion360 and revising hole diameters)
6. Open MatterControl
7. Reslice

**Expected Behavior**
The cache would be invalidated and the scene would be resliced

**Actual Behavior**
The cache is not invalidated and the original GCode with the incorrect mesh remains

I believe my repro holds but haven't stepped through it to verify and am simply reporting as experienced and based on what I understood at the time. For the first time ever I'm shuttling exported gcode onto memory sticks and although printing a small part, spend a bunch of time going back and forth. Yesterday I made about 3 or 4 back-to-back changes, without thinking assumed overwriting the file would work and kept getting the same printed part on the bed even though significant changes to the model occurred.

Not sure if you'd want to consider the linked files in the cache invalidation but not doing so requires the user to figure out why the output doesn't change, and if there's not enough variance to identify that the change didn't take place, will leave users in the dark.

jlewin, 2020-07-29:
Using the steps above another variation on this is as follows:

- Click the slice button to ensure the cache is primed for the current settings and file
- In the file system, replace the referenced mesh file with a significantly different model
- Click the slice button and note that nothing changes
- Be wise to the cache and change a single slice setting to invalidate the cache
- Click the slice button and observe the new mesh is now visible in the gcode view
- Revert the settings change to the original value
- Reslice

**Expected**
Ideally the current mesh would remain in the gcode view

**Actual**
The gcode incorrectly reverts to the original cache and model once the initial slice settings are restored

larsbrubaker, 2020-07-29:
Off the cuff I would say it should be very easy to add to the checksum the size of the referenced STL files.

I will look into adding that ASAP. Thanks for the detailed report. Hope your doing well.

jlewin, 2020-07-29:
In an ideal world, we'd compute the fingerprint for the file as the size may not change. Alternatively you might include the modified time on the file as a simple work around and handle the case where the content changed but the size did not.

larsbrubaker, 2020-07-29:
Checking the modified time of the visible files in the MCX file. Needs more testing but it is a start.

master #4807

closed 2020-08-02 by larsbrubaker