MatterHackers/MatterControl #4828: Library folder consistently reloads losing 'Generating' thumbnail icon and causing flicker

jlewin, 2020-08-29:
See this a lot while troubleshooting icon/theme issues. Wouldn't expect the listview to rebuild its children but doing so is the root o the problem. After some investigation it looks like the FileSystemWatcher is behind the reloads, and is firing off on LastAccess changed when we read in the file to generate the thumbnail.

Unclear why we're filtering for `NotifyFilters.LastAccess` event but doing so is behind the incorrect behavior and loss/incorrect representation of icon state and nasty flicker.

jlewin, 2020-08-29:
This issue was caused by watching for ` NotifyFilters.LastAccess`  and us invalidating and rebuilding folder content after the property change. By removing the `LastAccess` watch we aren't notified that directory content changed and function as expected

larsbrubaker, 2020-08-30:
Thanks for the fixe

closed 2020-08-30 by larsbrubaker