MatterHackers/MatterControl #4836: Cura "Lift head"

3ddruckermedartis, 2020-09-03:
Feature Request
===============

**Problem**
When minimal layer print time cannot be respected, print can become ugly because of too much heat in the part and therefore produce deformed prints.

**Solution**
Add a feature like the cura "lift head" feature so that the printer makes a pause in the print if the layer print time is not respected.

larsbrubaker, 2020-09-03:
We used to have this feature and it seemed incredibly destructive to the print, are you getting good results with this in cura?

3ddruckermedartis, 2020-09-08:
Actually I didn't use it. I get problems as soon as the layer print time is too low and reducing the speed is not the best solution. Different print speeds usually produce different print qualities and I want all the outer perimeter to print with the same speed indepedently from the layer print time. For the moment I have just one solution which is to print a tower, at th eopposite side of the bed, to force the layer print time to be longer.
Do you know the reason why this feature was destructive to the print ?