MatterHackers/MatterControl #4841: Not launching on Big Sur

onfoot, 2020-09-07:
Bug Report
============
**Steps to Reproduce**
1. Launch MatterControl on Big Sur beta
2. Yeah, that's it.

**Expected Behavior** Launch and use MatterControl

**Actual Behavior** Observing attached screen. When run from Terminal, the following is all that shows up:
```
./MatterControl.app/Contents/MacOS/MatterControlMac 
Time to '': 205
```

**MatterControl Build Number**
2.20.4.10455 ?

**Operating System Version**
macOS Big Sur Beta 20A5364e

Any way I can help debug this issue?

tuxotron, 2020-11-13:
I upgraded to Big Sur today and I'm having the same issue. Any solution so far?

onfoot, 2020-11-13:
Haven't tried since then. I bit the bullet and migrated my modeling to Blender. 😢

redonebe, 2020-11-16:
Same here. White screen in Big Sur.

joericki, 2020-11-16:
Also white screen. I contacted MatterHackers and got the response that they are flooded with support request and will answer personally in 2-3 days.

larsbrubaker, 2020-11-16:
Sorry for the delay, we are working on a new graphics system for Mac and Linux. Regrettably, the library we had been using on the Mac is no longer support.

We are quite close to having the new system working an anticipate it will be about a month until the new version is published.

larsbrubaker, 2020-12-14:
Update to any watching this issue:
We have the program up and running well. In fact it will be the best version on Mac ever. 
We are still having issues signing and packaging the distribution.
Hoping we can get an alpha up in the next few days.

willyan-arruda, 2020-12-14:
@larsbrubaker awaiting the functional version for big sur!

larsbrubaker, 2020-12-19:
This is fixed and the main download should be working.
There still an issue with signing (you have to go into settings to approve the installation)
It runs one core at 100%

There will be an update that fixes both of these next week.

willyan-arruda, 2020-12-19:
@larsbrubaker where can i find the main download?

closed 2020-12-21 by larsbrubaker