MatterHackers/MatterControl #4843: Terminal Visibility Options - Cannot suppress Ok responses

Wotsamatter, 2020-09-08:
Bug Report
============
**Steps to Reproduce**
1. Open Mattercontrol and connect to printer
2. Open the Terminal panel
3. Under Visibility Options, de-select 'Ok' Responses

**Expected Behavior**
The "ok <linenumber>" response messages should not appear, so the terminal view is not cluttered up.

**Actual Behavior**
The Visibility Option has no effect.  I get ok <linenumber> response messages scrolling up the terminal display constantly, regardless of visibility setting. The issue persists if I open/close MatterControl, or disconnect/reconnect the printer.  This causes wanted information in the terminal window to scroll out of sight after 10-20 seconds.

**MatterControl Build Number**
(To find this, go to the MatterControl menu in the top left and click "About MatterControl")
Version 2.20.9.10582

**Operating System Version**
Windows 10 Pro (2004)

**Printer Make/Model**
SeeMeCNC Rostock Max V2, Repetier firmware V0.91

larsbrubaker, 2020-09-10:
Would it be possible to attache the terminal log so that I can see the output that is not functioning?

Thanks.

Wotsamatter, 2020-09-10:
> <author@noreply.github.com>
> Cannot suppress Ok responses (#4843)

Hi Lars,

A sample terminal log file is attached.  I note that the terminal log 
file generally contains all send/receive responses, regardless of the 
visibility options view filtering in the Terminal viewer in the 
MatterControl user interface, and the sample attached contains what I 
would expect to see.

The issue I have reported only relates to the Visibility Options of the 
Terminal viewer in the UI.  If I disable other visibility options, such 
as Temperature Responses or Wait Reponses, these items don't appear in 
the viewer, however switching the visibility of the Ok responses seems 
to have no effect and these keep scrolling up the window.

Regards,
Lloyd.

Communication State: AttemptingToConnect
M105

N1 M110 N1*125

Communication State: Connected
N2 M114*37

Transformation matrix: 1.000000 0.000000 0.000000 0.000000 1.000000 0.000000 0.000000 0.000000 1.000000
Free RAM:1117
Autoretract:0
X:0.00 Y:0.00 Z:0.000 E:0.0000
SelectExtruder:0
FlowMultiply:100
ok
ok 2
X:0.00 Y:0.00 Z:0.000 E:0.0000
N3 M105*36

ok 3
T:0.00 /0 B:0.00 /0 B@:0 @:0
N4 M115*34

ok 4
FIRMWARE_NAME:Repetier_0.92.9 FIRMWARE_URL:https://github.com/repetier/Repetier-Firmware/ PROTOCOL_VERSION:1.0 MACHINE_TYPE:Delta EXTRUDER_COUNT:1 REPETIER_PROTOCOL:3
Printed filament:366.78m Printing time:2 days 6 hours 51 min
PrinterMode:FFF
wait
N5 M105*34

ok 5
T:19.12 /0 B:18.25 /0 B@:0 @:0
wait
N6 M105*33

ok 6
T:19.12 /0 B:18.25 /0 B@:0 @:0
wait
N7 M105*32

ok 7
T:19.12 /0 B:18.25 /0 B@:0 @:0
wait
N8 M105*47

ok 8
T:19.12 /0 B:18.25 /0 B@:0 @:0
wait
N9 M105*46

ok 9
T:19.12 /0 B:18.25 /0 B@:0 @:0
wait
[... 4205 lines cut ...]
wait
N897 M105*33

ok 897
T:19.12 /0 B:18.95 /0 B@:0 @:0
wait
N898 M105*46

ok 898
T:19.12 /0 B:18.95 /0 B@:0 @:0
wait
N899 M105*47

ok 899
T:19.12 /0 B:18.95 /0 B@:0 @:0
wait
N900 M105*46

ok 900
T:19.12 /0 B:18.95 /0 B@:0 @:0
wait
N901 M105*47

ok 901
T:19.12 /0 B:18.95 /0 B@:0 @:0

jlewin, 2022-12-21:
At first glance, it seems odd to have a trailing integer on the ok line, but given the context, I'll assume this format is correct and should be accounted for. The MatterControl behavior can be chalked up to explicitly testing the line for 'ok'  and can be circumvented by checking if the line starts with `ok` and adding some rules to guard against overriding other rules (like temperature lines that start with ok )

**Current**
[image]

Something like the following seems lightweight and allows for the trailing integer without stepping on other rules
[image]

Alternatively, a regular expression could specifically match this specific case but would have a larger overhead.