MatterHackers/MatterControl #4847: "Manually Configure Connection" COM Port selection vertical scrollbar needed

Marcus-Depew, 2020-09-09:
Feature Request
===============

**Problem**
Hard to pick a COM#Port when there's lots, more than ~10 ports 
**Solution**
Vertical scrollbar, to scroll from top of selection to bottom. 

[image]

krolco, 2021-01-28:
I built current source, and I'm pretty sure @larsbrubaker fixed this in this commit: c8bfef2085df31090f1c7d032cb15e63f411de4a
The window now has a drop-down for serial port, along with a list for baud rate:
[image]

larsbrubaker, 2021-01-28:
You are correct. The drop down can add a scroll bar if needed.

closed 2021-01-28 by larsbrubaker