MatterHackers/MatterControl #4854: unable to store software leveling

Tex-medic, 2020-09-14:
I am a unable to save a auto leveling settings, and now unable to print. Its keep requesting that I perform software leveling, and is not saving the data set. Perform The software leveling, and exit out of the setting, and it asks me to perform another software leveling, etc etc.

larsbrubaker, 2020-09-15:
Can you attach your printer settings so that we can try to debug this in context? Thank you.

[Here](https://www.matterhackers.com/mattercontrol/support/slice-settings/import-and-export?search=export) are some instructions if you need help exporting the settings. You will need to add them as a zip.

Tex-medic, 2020-09-15:
[Tex-Medic.zip](https://github.com/MatterHackers/MatterControl/files/5228751/Tex-Medic.zip)
Ok here is the zip file with the printer export, and 3 pics to include my printer confirming that the software level is complete, followed by the software again wanting to level itself again, and another of the level plots from the 1st 14 or 15 times I tried it solo. I have gone so far as to delete and re-install just the printer, and then to uninstall, and reinstall the software with the same results. I don't know if it will matter or not but I also just got done with some updates but kept installing the same one over and over again. I am at a total loss....

Ender 3 pro running stock mobo with stock OS with a BL touch all properly calibrated up until 2 days then this happened. No new hardware install, nada

Thanks

larsbrubaker, 2020-09-16:
Thanks for sending the file. I can see from the leveling data that there are multiple points that have the exact same x & y coordinates. This is an error that will trigger the need for re-leveling. It is likely that your printer cannot actually move to all the points and is hitting on of the edges.

Looking deeper I can see the probe offset is 44 in x. 
[image] 
This does not allow the current leveling solution to reach every part of the bed. Changing this to 22 will make the level complete.

[image]

[image]

I will keep this issue as we should really build the sampling grid with a consideration of the probe offset and that would fix this issue. For now you can just reduce the x probe offset.

Thanks for the report and the upload.

Tex-medic, 2020-09-16:
Ok but for devil's advocate that is the specs per Creality for the BL touch
with their stock mounting bracket. Why would it work with no issues then so
out of whack? I was even sitting here watching it run through the points
and the probe hit plate every single time, and I even reduced the size of
the printing plate from 235*235 to 220*220.

larsbrubaker, 2020-09-16:
I do not know how it could have worked correctly, but it does reproduce this error with the emulator so it should be easy for a developer to reproduce.

When we fix it, it will work with the correct value for the offset.

larsbrubaker, 2020-09-21:
fixed in the next build

closed 2020-09-21 by larsbrubaker