MatterHackers/MatterControl #4859: Start Gcode Disappearing

sf01425, 2020-09-16:
When ever I switch between my saved slicer settings, the Start gcode disappears and is replaced with G28. The End and Cancel gcode stays.

larsbrubaker, 2020-09-20:
If you export and upload your settings we can take a look at it. Thank you.

Trucky-89, 2020-10-01:
Hi Lars,

Stephen from MH here.  I am having the same issue with my personal printer [Printrbot], however, I do not experience this issue with the Pulse I have on loaner. When I select another material profile, only the Start Gcode section is reset to default, the other gcode fields are unaffected.

[Printrbot Metal Plus.zip](https://github.com/MatterHackers/MatterControl/files/5308791/Printrbot.Metal.Plus.zip)

larsbrubaker, 2020-10-07:
Here is the file that can be opened.

[Printrbot Metal Plus.zip](https://github.com/MatterHackers/MatterControl/files/5342938/Printrbot.Metal.Plus.zip)

larsbrubaker, 2020-10-07:
Looking at the profile, there should not be any 'Printer' settings set for any material or quality override.  I don't know how these got set, it may be from some build that allowed it or from some older profile. Either way, you probably should clear them so that there is not this confusion. If you need to change a 'Printer' setting for a given material or quality setting we should look at that as a separate issue.

I have included a quick gif on clearing these settings.

[image]

closed 2020-10-16 by larsbrubaker