MatterHackers/MatterControl #4879: Auto bed level failure

Tex-medic, 2020-10-12:
Ok since the last update I have been unable to perform a bed level. The software is not allowing for BL touch offset when probing. No matter what I put in for bed size or plate size it when to goes to probe on the left side of the plate the print nozzle lines up but the BL touch is completely off the map. Any ideas?

Tex-medic, 2020-10-19:
Here are some pics I took of the issue. 
[image]
[image]
[image]

One pic shows how far over the extruder goes, and the other shows what the computer is doing for grid leveling. Any ideas?

larsbrubaker, 2020-10-25:
Can you please export your profile and attach it as a zip to this issue. That will help us look into what is happening.

The first place that I would look to make sure you have good data is in the probe offset. Looking at the image you sent I would expect there to be a negative 30 or 40 in the x.

[image]

closed 2020-10-30 by larsbrubaker