MatterHackers/MatterControl #4880: terminal selection issues

phazei, 2020-10-17:
Feature Request
===============

**Problem**
Unable to select text in the terminal window to copy a single line or few.  If there's an output I need to copy I have to export the entire log, then search through it for the line I wanted.

**Solution**
Selectable text would make the terminal much more functional and simpler to use.

I find your terminal to be one of the best of the choices out there.  The visibility options are great and it's dark themed to boot.  Being able to select text from it would be great.

aahutsal, 2022-05-14:
@larsbrubaker is it still actual?

larsbrubaker, 2022-05-15:
If you're asking if this is still an issue, or if this bounty is still active, then both are yes. We would still like to be able to select text in the terminal and this bounty is still active.

aahutsal, 2022-05-15:
@larsbrubaker thanks. I'll take a look

aahutsal, 2022-05-17:
@larsbrubaker I'd inherit `TextScrollWidget` from TextBox or even replace with `TextBox`. Would start if you have no problems with that

aahutsal, 2022-05-28:
@larsbrubaker could you please just hint me if I'm on the right path?

larsbrubaker, 2022-05-30:
@aahutsal I did not realize you were looking for feedback. I do not know if that would be an effective approach. My theory would be that the existing control needs to have consideration for selection and not be transformed into an edit control. 

That said, I have not delved into this problem in a compelling way and there may be a different better solution.

phazei, 2024-01-30:
Any update on this?  I saw the merged PR but it didn't seem to do anything that I could tell