MatterHackers/MatterControl #4907: Printer Speed Slider

Marcus-Depew, 2020-11-13:
Feature Request
===============

**Problem**
Print speeds must be set before every print.
**Solution**
Print speed slider to change print speeds in real time.

larsbrubaker, 2020-11-13:
Does this not do what you want?

[image]

Marcus-Depew, 2020-11-13:
This was a request from @ImprezaRSC, they are trying to test different print speeds quickly. 

Maybe it would be best to have one sliders for each on of these categories that changes the speed in unison? 

[image]

larsbrubaker, 2020-11-13:
This is part of the slicing engine and the output gcode and cannot be controlled in real-time without a lot of work. I will add it to the wishlist.

ImprezaRSC, 2020-11-13:
Fair enough. Thank you. I'm just getting comfortable and it seemed like a good idea.

Thanks,

Gary