MatterHackers/MatterControl #4916: macOS Big Sur, White/Blank Screen

Marcus-Depew, 2020-11-18:
Bug Report
============
**Steps to Reproduce**
Install and Launch MatterControl 2.20.4.10454 on MacOS Big Sur

**Actual Behavior**
Blank/White screen, resizing has no effect.
[image]

This is the same version that runs on Catalina. Appdata folder was also cleared for good measure.  

**MatterControl Build Number**
2.20.4.10454
**Operating System Version**
macOS 11 Big Sur

Marcus-Depew, 2020-11-18:
BTW, this screenshot was taking over VNC, but the same error happens on a really monitor.

larsbrubaker, 2020-11-20:
dupe of #4841 Not launching on Big Sur

closed 2020-11-20 by larsbrubaker