MatterHackers/MatterControl #4947: Subtract Erratic

taijireality, 2021-01-07:
Bug Report
============
**Steps to Reproduce**
1. I've got a pyramid that I hollowed out by subtracting a 90% smaller pyramid.
2. I'm then taking cubes to cut out openings in the sides of the hollowed out pyramid also using subtract.

**Expected Behavior**

I expect the openings in the sides of the pyramid to correspond with the subtracted cubes.

**Actual Behavior**

Sometimes the subtract function hangs - sometimes it appears to complete but nothing happens and sometimes it works partially creating unexpected artifacts.  

Here's a video demonstrating the issue:

https://user-images.githubusercontent.com/39426654/103853655-7a5f9180-5063-11eb-8f27-a42466180c09.mp4

**MatterControl Build Number**
2.20.11.10793
(To find this, go to the MatterControl menu in the top left and click "About MatterControl")

**Operating System Version**
Mac OS 11.1

**Printer Make/Model**
n/a

67Exec, 2021-01-08:
I am having a similar issue wherein it will either remove both items from the work space, or exit the program, generally the latter. I am trying to subtract the red cube in order to remove a portion of the model on both sides as indicated by the green line I've drawn opposite of the cube. 
I'd be happy to share the file if needed
[image]
**OS:**
Win 10 Pro 1909 build 18363.1198

**MC:** 
2.20.11.10798

mlewis-everley, 2021-06-21:
I am getting a very similar issue, I can subtract one or two items from a shape, but when I try to subtract any more I get the "subtract" progress bar appear in the status bar at the bottom of the screen, but it seems to hang and never completes. 

**OS:**
Linux Mint 20

**MC:**
2.19.10.10370 (there does not appear to be a newer binary than this for Linux)

larsbrubaker, 2022-04-20:
This should now be fixed

closed 2022-04-20 by larsbrubaker