MatterHackers/MatterControl #4950: Check For Update hangs when downloading

taijireality, 2021-01-09:
Bug Report
============
**Steps to Reproduce**
1. All attempts to download update from within MatterControl hang
2. 
3. 

**Expected Behavior**

a downloadable update

**Actual Behavior**

typically hangs at around 10-20%

**MatterControl Build Number**
2.20.11.10793
(To find this, go to the MatterControl menu in the top left and click "About MatterControl")

**Operating System Version**
Mac OS 11.1

**Printer Make/Model**
n/a

larsbrubaker, 2021-01-09:
Thank you for the feedback. This issue became a problem because we switch from .dmg to .pkg. Regrettably there was no way to fix the build you are having trouble with. You (any everyone else) will need to download from the download page and re-install. 

It has been fixed in the new build and should not be a problem with future updates.

Thanks again.

taijireality, 2021-01-10:
Thanks for the email!  I think you've got a great piece of software under
development.

I'm presently stuck on my present project until subtract is working again.

When the kinks gets ironed out I hope to be able to purchase a subscription
down the road.

Thanks again,

Noah W.
http://taijireality.substack.com

closed 2021-01-11 by larsbrubaker

taijireality, 2021-02-07:
Update still not working correctly.  The package downloads but then I get
the following:

[image: image.png]