MatterHackers/MatterControl #4953: Multiple Objects in 1 STL

mstange42, 2021-01-12:
Bug Report
============

**Expected Behavior**
Parse/Slice Correctly

**Actual Behavior**
In the case Im supplying, instead of doing a union/addition, it does a difference/subtraction. There are also case where it does other weird things I just don't have any handy. 

**MatterControl Build Number**
All

[image]
[Test.zip](https://github.com/MatterHackers/MatterControl/files/5799201/Test.zip)

larsbrubaker, 2021-01-12:
This has lots of self intersecting geometry which can have an undefined result. It is hard for the algorithm to find a solution. I tried to repair it with [NetFad](https://service.netfabb.com/service.php) and it gave an error. I would recommend you try to repair it with another mesh tool (possibly meshmixer or meshlab. Or if you can re-model it.

I also tried [this ](https://www.formware.co/onlinestlrepair) online tool and it produced [Test_fixed.zip](https://github.com/MatterHackers/MatterControl/files/5799308/Test_fixed.zip) as a  result.

Here is a screenshot of the fixed mesh.
[image]

We will keep this as a test model (if its ok) to try and improve the algorithm, but it seems like this may not be fixable.

mstange42, 2021-01-12:
The easy thing to do, is just separate the objects. If i load them
individually it works as intended. It is no different from dropping 2
primatives and putting them together so they intersect.

larsbrubaker, 2021-01-22:
Again, thanks for submitting this issue. The next release of MatterControl will correct this for this model.

[image]