MatterHackers/MatterControl #4955: OSX: Error when updating beta

salamanders, 2021-01-21:
I'm on the beta channel.  I click update.  It downloads the update.  But it errors instead of applying the update.

* Current Build: 2.20.11.10793
* Mac OS 10.15.7
* Update Channel: Beta
* Reproducibility: 4 of 4 tries.

Marcus-Depew, 2021-01-21:
Thank you for the info. 

The best option in the meantime is to uninstall normally and download this version and install it: https://mattercontrol.appspot.com/downloads/mattercontrol-mac-os-x/development

larsbrubaker, 2021-01-21:
Regrettably the December build for Mac specified that the installer was a .dmg rather than a .pkg and so did not run it correctly. This is fixed and will not happen with future builds. As indicated, you will need to re-download and update to get it working.

Thanks Marcus.

salamanders, 2021-01-21:
Will do, thanks for the workaround!

closed 2021-01-21 by salamanders