MatterHackers/MatterControl #4959: Open Recent does not refresh correctly on first try

larsbrubaker, 2021-01-25:
@larsbrubaker commented on [Sat Dec 05 2020](https://github.com/MatterHackers/MCCentral/issues/6193)

[image]

jlewin, 2021-02-01:
In my opinion, the implementation that was in place in late 2019 was a quite a bit cleaner than the revamped code.

Having said that, the problem that exists in both cases appears to be something that never repro'd on my machine. From the screencast above and from troubleshooting on my machine, the unexpected white ImageBuffers are caused not by the thumbnail generation logic (tracing and dumping generated images shows nothing but valid and expected content) but rather something that happens in Agg. 

Specifically, ImageGlPlugin.Remove(imageBuffer) is called on the ImageBuffer before it is updated with new content and it was assumed that the byte array would uniquely identify and dump the old texture. What ends up happening at runtime is the
ImageGlPlugin.Remove(imageBuffer) call queues a delete, the new image data is assigned, and a few cycles later the texture is deleted and somewhere along the way we're back to white pixels.

With the build that I'm on (circa July 2019) if I comment out the `ImageGlPlugin.Remove(imageBuffer);` call and put in `imageBuffer.MarkImageChanged();` I get what I originally saw the last time I worked on this:

**Quick Revision**
[image]

**Updated Rendering Behavior**
[image]

*This quick change seems to work great but I believe ends up leaking textures*

Someone who's interested might look into or wire up a better leak free version of updating the ImageBuffer and freeing the Texture.

larsbrubaker, 2021-02-01:
Good catch. 

Don't know where that code originally came from. The GL cache should not be leaking without the delete. The Mark should free the memory.

larsbrubaker, 2021-02-01:
Fixed

larsbrubaker, 2021-02-01:
That definitely counts as a fix if you want to claim it on bounty source.

closed 2021-02-01 by larsbrubaker

jlewin, 2021-02-01:
I wrote it but I swear you and I originally reviewed and couldn't come up with a way to purge the old image data and refresh without a leak that we were seeing. You should definitely delete the `ImageGlPlugin.Remove` implementation as well, if it won't be used in the case, as it was added just for this. I'm sure a test could prove if a leak happens as I remember but without one it's just hearsay.