MatterHackers/MatterControl #4970: Zoom to cursor position in 3D view

larsbrubaker, 2021-02-03:
In the 3D view make it so that zooming in and out is sensitive to the object under the cursor and zooms to the position of the closest 3D object being pointed to. If there is no object under the cursor zoom to a spot that is consistent with the scene and will not have the camera zooming off to nowhere.

Here is an example of zooming to the cursor:
[image]

Here is a site that does a good job (this is where the sample came from).

https://thangs.com/model/10814

visdauas, 2021-03-13:
Hey, is this bounty still active?

ishapiro, 2021-03-14:
I assume so.  I never did anything to withdraw it.

larsbrubaker, 2021-03-14:
Yes, this bounty is still active.

larsbrubaker, 2021-03-22:
The submitted pull request does the zooming behavior, but after zooming the normal pan, zoom and rotate controls do not operate well.

An acceptable fix must leave the view in a good state. The fix needs to include moving the origin and eye position of the world transform so that rotate, pan and zoom still operate in a reasonable manner.

PR #5012 (2021-04-15): Added zoom to mouse position
Fixes #4970 
It works as requested if the mouse is over nothing it zooms to the center, otherwise it zooms exactly to the mouse cursor.
You can modify the zoom amount with `View3DWidget.ZoomDelta`. Only tested on windows.
closed 2021-04-15 by larsbrubaker