MatterHackers/MatterControl #4988: Failing to load on macOS 11.2.1

ishapiro, 2021-02-22:
I previously had an earlier version of MatterControl that stopped working when I upgrade to macOS 11.   I downloaded the latest build, MatterControl 2.21.2 (February 5, 2021), and install it.

When starting the app the icon bounces a couple of times, then the screen flashes with the MatterControl APP, and then the app aborts.

The following message is in system.log on the mac:

Feb 22 11:08:55 DrVAX-iMac MatterControlMac[1316]: getattrlist failed for /System/Library/Frameworks/OpenGL.framework/Resources//GLRendererFloat.bundle/GLRendererFloat: #2: No such file or directory
Feb 22 11:08:55 DrVAX-iMac com.apple.xpc.launchd[1] (application.com.MatterHackers.MatterControl.12983087210.12983087216[1316]): Service exited with abnormal code: 1

ishapiro, 2021-02-25:
I have highlighted MatterControl on my YouTube channel and believe multi-platform support, specifically macOS support is key to its success.  I am a software engineer by training and happy to work on testing but do not have C# skills to debug this issue at the code level.

Marcus-Depew, 2021-03-23:
Hello, I just tested it with the very latest version of [MatterControl (2.21.4.10914)](https://mattercontrol.appspot.com/downloads/development/ag9zfm1hdHRlcmNvbnRyb2xyOwsSB1Byb2plY3QY7AcMCxINUHVibGljUmVsZWFzZRiAgPCo5NeBCwwLEgZVcGxvYWQYgIDwqKzqlQgM). 

Try this version, but  if it still doesn't load use the below command to launch it from the terminal and post the error message if there is one: 

$ open /Applications/MatterControl.app 

I am curious to know what happens!

ishapiro, 2021-03-23:
Thanks.  I will try it now.

Irv

- - - - - - - - -

[image: photo]
*Irv Shapiro | Personal Email*
Cogitations, Inc.

Mobile: 847 331 1310 | ***@***.***

http://www.cogitations.com
<https://mailtrack.io/trace/link/a763772352764d038f2077e1428b6de512cdcc40?url=http%3A%2F%2Fwww.cogitations.com&userId=2835845&signature=a290c065d4525d4c>
| Skype: ishapiro55 <#SignatureSanitizer_SafeHtmlFilter_>
<https://mailtrack.io/trace/link/5cc51cef44fcdb2817601d45205fb190a58e797f?url=http%3A%2F%2Fus.linkedin.com%2Fin%2Firvshapiro&userId=2835845&signature=9da4d19232481f4b>
<https://mailtrack.io/trace/link/03b83a58f4cb39f2976f363e9d94653b8fa417c3?url=http%3A%2F%2Ftwitter.com%2Firvshapiro&userId=2835845&signature=61afa765f28a06a6>

ishapiro, 2021-03-23:
Marcus,

Marcus,

I just tried the version you included in the email.  Basically, the icon
bounces in the dock, the logo displays on the screen, and then the app
exits.

I have linked a screen recording and attached the macOS console log
showing the errors.  The behavior is the same when I open from a terminal
using the app path or launch from an icon.

https://www.icloud.com/iclouddrive/00SNBk1FDbk-Wlt3QT0E9MstA#mattercontrol_starup_mac
<https://mailtrack.io/trace/link/3a688eb689ed9957fe3c636f566b754f36e49636?url=https%3A%2F%2Fwww.icloud.com%2Ficlouddrive%2F00SNBk1FDbk-Wlt3QT0E9MstA%23mattercontrol_starup_mac&userId=2835845&signature=730496229fd55281>

When I do a show package content and then open the app directly in a
terminal I get the following output:

Last login: Tue Mar 23 17:33:27 on ttys000

/Applications/MatterControl.app/Contents/MacOS/MatterControlMac ; exit;

irvshapiro@DrVAX-iMac ~ %
/Applications/MatterControl.app/Contents/MacOS/MatterControlMac ; exit;

Time to '': 465

Time to 'ShowAsSystemWindow': 1

Time to 'First draw->RunOnIdle': 0

Time to 'MatterControlApplication.Initialize': 0

Time to 'Loading...': 0

Time to 'Loading...': 0

Time to 'Loading...': 0

Time to 'Loading...': 0

Time to 'Loading...': 304

2021-03-23 17:41:33.965 MatterControlMac[49647:4525512]

Unhandled Exception:

GLFW.Exception: Cocoa: Failed to find service port for display

  at GLFW.Glfw.GlfwError (GLFW.ErrorCode code, System.IntPtr message)
[0x0000b] in <710ef59769dd43cebbe8cbc1191b938a>:0

  at (wrapper native-to-managed) GLFW.Glfw.GlfwError(GLFW.ErrorCode,intptr)

  at (wrapper managed-to-native) GLFW.Glfw.PollEvents()

  at MatterHackers.GlfwProvider.GlfwPlatformWindow.Show () [0x003ce] in
<4657ea20249b45e391f8c9e4d9effa7b>:0

  at MatterHackers.GlfwProvider.GlfwPlatformWindow.ShowSystemWindow
(MatterHackers.Agg.UI.SystemWindow systemWindow) [0x00021] in
<4657ea20249b45e391f8c9e4d9effa7b>:0

  at MatterHackers.Agg.UI.SingleWindowProvider.ShowSystemWindow
(MatterHackers.Agg.UI.SystemWindow systemWindow) [0x00394] in
<0a373763f2754fd99f5d7dc4f7d4c936>:0

  at MatterHackers.GlfwProvider.GlfwWindowProvider.ShowSystemWindow
(MatterHackers.Agg.UI.SystemWindow systemWindow) [0x00026] in
<4657ea20249b45e391f8c9e4d9effa7b>:0
[... 159 lines cut ...]
*Irv Shapiro | Personal Email*
Cogitations, Inc.

Mobile: 847 331 1310 | ***@***.***

http://www.cogitations.com
<https://mailtrack.io/trace/link/a926be3e90cdd160d309d21ecc14ed6079f114f3?url=http%3A%2F%2Fwww.cogitations.com&userId=2835845&signature=dda6cf46ef9e93d0>
| Skype: ishapiro55 <#SignatureSanitizer_SafeHtmlFilter_>
<https://mailtrack.io/trace/link/560aa382cc1ae5528fa38ce011f2a4fb8a16ad2a?url=http%3A%2F%2Fus.linkedin.com%2Fin%2Firvshapiro&userId=2835845&signature=cbc179d993a263df>
<https://mailtrack.io/trace/link/cf92bdf3011eacbb811fda64bd9520c4c9046e80?url=http%3A%2F%2Ftwitter.com%2Firvshapiro&userId=2835845&signature=f5b2560a9d66eec7>

Mar 23 17:30:39 DrVAX-iMac MatterControlMac[49330]: getattrlist failed for /Library/GPUBundles/AMDRadeonX4000GLDriver.bundle/Contents/MacOS/ATIRadeonX4000SCLib.dylib: #2: No such file or directory
Mar 23 17:30:39 DrVAX-iMac MatterControlMac[49330]: getattrlist failed for /System/Library/Extensions/AMDRadeonX4000GLDriver.bundle/Contents/MacOS/ATIRadeonX4000SCLib.dylib: #2: No such file or directory
Mar 23 17:30:39 DrVAX-iMac MatterControlMac[49330]: getattrlist failed for /System/Library/Frameworks/OpenGL.framework/Resources//GLRendererFloat.bundle/GLRendererFloat: #2: No such file or directory
Mar 23 17:30:40 DrVAX-iMac com.apple.xpc.launchd[1] (application.com.MatterHackers.MatterControl.12993977885.12993977891[49330]): Service exited with abnormal code: 1
Mar 23 17:31:42 DrVAX-iMac com.apple.xpc.launchd[1]: Coalition Cache Hit: app<application.com.MatterHackers.MatterControl.12993977885.12993977891(501)> [154343]
Mar 23 17:31:45 DrVAX-iMac MatterControlMac[49350]: getattrlist failed for /Library/GPUBundles/AMDRadeonX4000GLDriver.bundle/Contents/MacOS/ATIRadeonX4000SCLib.dylib: #2: No such file or directory
Mar 23 17:31:45 DrVAX-iMac MatterControlMac[49350]: getattrlist failed for /System/Library/Extensions/AMDRadeonX4000GLDriver.bundle/Contents/MacOS/ATIRadeonX4000SCLib.dylib: #2: No such file or directory
Mar 23 17:31:45 DrVAX-iMac MatterControlMac[49350]: getattrlist failed for /System/Library/Frameworks/OpenGL.framework/Resources//GLRendererFloat.bundle/GLRendererFloat: #2: No such file or directory
Mar 23 17:31:45 DrVAX-iMac com.apple.xpc.launchd[1] (application.com.MatterHackers.MatterControl.12993977885.12993977891[49350]): Service exited with abnormal code: 1
Mar 23 17:33:31 DrVAX-iMac com.apple.xpc.launchd[1]: Coalition Cache Hit: app<application.com.MatterHackers.MatterControl.12993977885.12993977891(501)> [154343]
Mar 23 17:33:33 DrVAX-iMac MatterControlMac[49408]: getattrlist failed for /Library/GPUBundles/AMDRadeonX4000GLDriver.bundle/Contents/MacOS/ATIRadeonX4000SCLib.dylib: #2: No such file or directory
Mar 23 17:33:33 DrVAX-iMac MatterControlMac[49408]: getattrlist failed for /System/Library/Extensions/AMDRadeonX4000GLDriver.bundle/Contents/MacOS/ATIRadeonX4000SCLib.dylib: #2: No such file or directory
Mar 23 17:33:33 DrVAX-iMac MatterControlMac[49408]: getattrlist failed for /System/Library/Frameworks/OpenGL.framework/Resources//GLRendererFloat.bundle/GLRendererFloat: #2: No such file or directory
Mar 23 17:33:33 DrVAX-iMac com.apple.xpc.launchd[1] (application.com.MatterHackers.MatterControl.12993977885.12993977891[49408]): Service exited with abnormal code: 1

ishapiro, 2021-05-21:
I just tried version 2.21.5 on and it is now working on Big Sur.  I will test on additional machines over the weekend.

larsbrubaker, 2021-05-21:
Yes, this was fixed on 5-14 (by me)

ishapiro, 2021-05-21:
Excellent.  Thanks.

Irv

- - - - - - - - -

[image: photo]
*Irv Shapiro | Personal Email*
Cogitations, Inc.

Mobile: 847 331 1310 | ***@***.***

http://www.cogitations.com
<https://mailtrack.io/trace/link/4218d286595027a76febb98e420287755b270ab8?url=http%3A%2F%2Fwww.cogitations.com&userId=2835845&signature=7240e5b6de869ee0>
| Skype: ishapiro55 <#SignatureSanitizer_SafeHtmlFilter_>
<https://mailtrack.io/trace/link/cdcdd0daf8a73fef6e60f675f999af578fabeabc?url=http%3A%2F%2Fus.linkedin.com%2Fin%2Firvshapiro&userId=2835845&signature=6717c57f349f8125>
<https://mailtrack.io/trace/link/589b904ca65c818ec8150e94224fce9472a9965f?url=http%3A%2F%2Ftwitter.com%2Firvshapiro&userId=2835845&signature=7587900afd43f188>

ishapiro, 2021-05-24:
Verified on a second Mac, different model, with Big Sur.  Works correctly.  Closing issue.

ishapiro, 2021-05-24:
Works correctly.

closed 2021-05-24 by ishapiro