MatterHackers/MatterControl #4994: Ampersand in text Object

rlcrom, 2021-02-25:
I'm not sure that this is an issue with MatterControl or Cura, however, if I use the text object and use an ampersand (bill & Diane) and export as an AMF file the file cannot be opened by Cura. It says it corrupt. If I export as a stl, it's okay in cura.

Marcus-Depew, 2021-03-23:
I'd like to try and replicate this, what OS, and versions of MatterControl and Cura are you using?

rlcrom, 2021-03-23:
Windows 10, cura 4.8.0  Mattercontrol 2.20.11 is what I'm using now. As I upgraded to their current version and the export was brokem. I just tried this again and it fails to load in Cura

larsbrubaker, 2021-03-23:
There does seems to be a problem with the AMF export and conformance to the AMF standard. There is very little documentation on valid AMF files and there is no validator that I am aware of. We will attempt to repair this in the near future and test against some know AMF loaders such as Cura, Blender, Mesh Mixer, etc...

BTW The export bug has been fixed in the current version.