MatterHackers/MatterControl #5000: Move the "Export" function for .gcode exporting

MichaelPetitclerc, 2021-03-03:
Feature Request
===============

**Problem**
Some users are not finding the export option that is currently under the "Bed" icon in the top bar and as such are trying to use the "Save As" button to export prints. 

**Solution**
Moving the "Export" function to be more visible and in the top bar will reduce confusion and reduce incoming requests for support with "File exported from MC is not visible on the printer's touchscreen" type issues.

larsbrubaker, 2021-03-04:
We are expanding the capabilities of the export under the 'Print' menu. Does that satisfy what you are needing here?

[image]

MichaelPetitclerc, 2021-03-04:
That could resolve it for some but the core problem is for customers who are not intrinsically looking to print the file but export it so that location may not be in the list of options they would use.  Possibly adding an option of "Export" in the "Save" dropdown with "Save As" that would open up the directory from the standard "Export" button.  

Most users who run into this issue are thinking of "Saving" their work as a file, much like a word doc.

larsbrubaker, 2021-03-04:
Adding export to the save drop down sounds like a solid and easy way to try and address the issue. 

Will add it and we can check back in a few months if there continues to be customer confusion.

Thank you.

larsbrubaker, 2021-03-05:
This will be in the next build:

[image]

closed 2021-03-11 by larsbrubaker