MatterHackers/MatterControl #5032: Make orthographic view mode

larsbrubaker, 2021-04-15:
@larsbrubaker commented on [Sat Aug 26 2017](https://github.com/MatterHackers/MCCentral/issues/1870)

- [ ] Modify the TrackballTumbleWidgetExtended to create the effect
- [ ] Add a button to enable / disable the mode (placed in menu with home, shading, bed)
- [ ] Animate the transition to and from orthographic (use the tumble cube as a reference how to animate the view)

Proposed Method:
- Pull the camera a great distance away from the scene
- Put the near and far clipping planes around the scene to get the correct contents
- Rotation should work normally
- Translation should work normally
- Scaling will require the view frustum to change its width depending on the zoom

fortsnek9348, 2022-02-24:
> Pull the camera a great distance away from the scene
> Put the near and far clipping planes around the scene to get the correct contents

Does this mean that the orthographic view should effectively put the camera at an infinite distance from the scene at all times? And how would the position of the perspective camera be determined when switching back? Remember the original position? Reposition such that the scene maintains a similar on-screen size?

larsbrubaker, 2022-02-24:
This issue is a dupe of #5187 and it should not exist. Only the bounty with the higher payout should exist.

Closing

closed 2022-02-24 by larsbrubaker