MatterHackers/MatterControl #5038: Pan viewing control by holding CRTL or ALT

Trucky-89, 2021-04-20:
Feature Request
===============

**Problem**
For users without a traditional mouse with an included scroll wheel (ie. trackball users), it is impossible to "pan" the viewing angle within the build platform.

**Solution**
Allow for the "pan" viewing feature when holding either CTRL or ALT in conjunction with using the right mouse button (normally used to rotate the viewing angle).

larsbrubaker, 2021-04-20:
This is already present. Let us know if it is not working correctly.

[image]

larsbrubaker, 2021-04-21:
Here's the right picture:

[image]

Trucky-89, 2021-04-21:
Thank you so much!  I was using the right mouse button by mistake. Works
like a charm!

closed 2021-04-21 by larsbrubaker