MatterHackers/MatterControl #5085: Newest Version MatterControl Crashing

Hal1949, 2021-06-18:
Bug Report
============
**Steps to Reproduce**
1. Load in *.obj file
2. UNGROUP
3. 

**Expected Behavior**
To UNGROUP an *.obj file.  In the previous version I could UNGROUP without any issue.
 
**Actual Behavior**
MatterControl stops working. and this window pops up.  ERROR "Unhandled exception has occurred in you application."  So, I click on the QUIT button.  I can repeat this error every time.

[image]

**MatterControl Build Number**
(To find this, go to the MatterControl menu in the top left and click "About MatterControl")
Upgrade version: 2.21.6.11058
[image]

**Operating System Version**
Windows 10
[image]

**Printer Make/Model**
GEEETech Pro i3 and Creality Ender 3 V2

larsbrubaker, 2021-06-18:
We will take this on as as ASAP required fix. Can you add the .obj file so we make sure we have the exact error you are seeting?

Thanks.

Hal1949, 2021-06-19:
I found a way around the problem last night.   If I export it as an *.STL file then reload the *.STL file it will UNGROUP.   I can deal with that but maybe it could be noted someplace so it is known to all.   I can't send a copy of the *.OBJ file I used because it's an original design and it isn't finished, and I don't have permission to share (Sorry).  The error will happen with any *.OBJ file.

larsbrubaker, 2021-06-19:
We will verify and fix the issue. 

Thank you.

larsbrubaker, 2021-06-24:
Fixed. Will be in the next build.

closed 2021-06-24 by larsbrubaker