MatterHackers/MatterControl #5096: mscorlib.dll missing after trying to run mattercontrol (Mono installed)

smblackledge, 2021-07-22:
Installed both Mono and MatterControl.  A nasty little error occurred, mscorlib.dll file not found.

**Operating System Version**
Linux 20.04

$ dpkg -l mono-complete
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version                           Architecture Description
+++-==============-=================================-============-==========================================================
ii  mono-complete  6.12.0.122-0xamarin1+ubuntu2004b1 amd64        complete Mono runtime, development tools and all libraries

$ dpkg -l mattercontrol
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version       Architecture Description
+++-==============-=============-============-=================================
ii  mattercontrol  2.19.10.10367 all          Software solution for 3D printers

$ which mattercontrol
/usr/bin/mattercontrol
dio@dio3dpsrv:~$ mattercontrol
Time to '': 228
exception inside UnhandledException handler: (null) assembly:/usr/lib/mono/4.5/mscorlib.dll type:FileNotFoundException member:(null)

[ERROR] FATAL UNHANDLED EXCEPTION: System.IO.FileNotFoundException: Could not load file or assembly 'Microsoft.VisualBasic, Version=10.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a' or one of its dependencies.
File name: 'Microsoft.VisualBasic, Version=10.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a'
  at Mindscape.Raygun4Net.RaygunMessageBuilder.SetEnvironmentDetails () [0x0000b] in <f91f2b249ee243acbec1dd6bfd729d19>:0
  at Mindscape.Raygun4Net.RaygunClient.BuildMessage (System.Exception exception, System.Collections.Generic.IList`1[T] tags, System.Collections.IDictionary userCustomData, Mindscape.Raygun4Net.Messages.RaygunIdentifierMessage userInfoMessage, System.Nullable`1[T] currentTime) [0x00024] in <f91f2b249ee243acbec1dd6bfd729d19>:0
  at Mindscape.Raygun4Net.RaygunClient.StripAndSend (System.Exception exception, System.Collections.Generic.IList`1[T] tags, System.Collections.IDictionary userCustomData, Mindscape.Raygun4Net.Messages.RaygunIdentifierMessage userInfo, System.Nullable`1[T] currentTime) [0x00016] in <f91f2b249ee243acbec1dd6bfd729d19>:0
  at Mindscape.Raygun4Net.RaygunClient.Send (System.Exception exception, System.Collections.Generic.IList`1[T] tags, System.Collections.IDictionary userCustomData, Mindscape.Raygun4Net.Messages.RaygunIdentifierMessage userInfo) [0x00023] in <f91f2b249ee243acbec1dd6bfd729d19>:0
  at Mindscape.Raygun4Net.RaygunClient.Send (System.Exception exception) [0x00000] in <f91f2b249ee243acbec1dd6bfd729d19>:0
  at MatterHackers.MatterControl.Program+<>c.<Main>b__6_2 (System.Object s, System.Threading.ThreadExceptionEventArgs e) [0x0001c] in <e58f04b6857344df9781155f8ee2c0c8>:0
  at System.Windows.Forms.Application.OnThreadException (System.Exception t) [0x00037] in <6d635ac3dc1c4424ad385ded79f1e868>:0
  at System.Windows.Forms.NativeWindow.OnThreadException (System.Exception e) [0x00000] in <6d635ac3dc1c4424ad385ded79f1e868>:0
  at System.Windows.Forms.NativeWindow.WndProc (System.IntPtr hWnd, System.Windows.Forms.Msg msg, System.IntPtr wParam, System.IntPtr lParam) [0x00196] in <6d635ac3dc1c4424ad385ded79f1e868>:0
  at System.Windows.Forms.XplatUIX11.SendMessage (System.IntPtr hwnd, System.Windows.Forms.Msg message, System.IntPtr wParam, System.IntPtr lParam) [0x000d5] in <6d635ac3dc1c4424ad385ded79f1e868>:0
  at System.Windows.Forms.XplatUIX11.CreateWindow (System.Windows.Forms.CreateParams cp) [0x004e5] in <6d635ac3dc1c4424ad385ded79f1e868>:0
  at System.Windows.Forms.XplatUI.CreateWindow (System.Windows.Forms.CreateParams cp) [0x00000] in <6d635ac3dc1c4424ad385ded79f1e868>:0
  at System.Windows.Forms.NativeWindow.CreateHandle (System.Windows.Forms.CreateParams cp) [0x00009] in <6d635ac3dc1c4424ad385ded79f1e868>:0
  at System.Windows.Forms.Control.CreateHandle () [0x00031] in <6d635ac3dc1c4424ad385ded79f1e868>:0
  at System.Windows.Forms.Control.CreateControl () [0x00039] in <6d635ac3dc1c4424ad385ded79f1e868>:0
  at (wrapper remoting-invoke-with-check) System.Windows.Forms.Control.CreateControl()
  at System.Windows.Forms.Control.CreateControl () [0x00072] in <6d635ac3dc1c4424ad385ded79f1e868>:0
  at System.Windows.Forms.Control.SetVisibleCore (System.Boolean value) [0x00057] in <6d635ac3dc1c4424ad385ded79f1e868>:0
  at System.Windows.Forms.Form.SetVisibleCore (System.Boolean value) [0x00065] in <6d635ac3dc1c4424ad385ded79f1e868>:0
  at MatterHackers.Agg.UI.WinformsSystemWindow.SetVisibleCore (System.Boolean value) [0x00009] in <712ffdb4c0144e8c88c0ce33a0b20977>:0
  at System.Windows.Forms.Control.set_Visible (System.Boolean value) [0x00009] in <6d635ac3dc1c4424ad385ded79f1e868>:0
  at System.Windows.Forms.Control.Show () [0x00000] in <6d635ac3dc1c4424ad385ded79f1e868>:0
  at MatterHackers.Agg.UI.WinformsSystemWindow.Show () [0x000f8] in <712ffdb4c0144e8c88c0ce33a0b20977>:0
  at MatterHackers.Agg.UI.WinformsSystemWindow.ShowSystemWindow (MatterHackers.Agg.UI.SystemWindow systemWindow) [0x00060] in <712ffdb4c0144e8c88c0ce33a0b20977>:0
  at MatterHackers.Agg.UI.SingleWindowProvider.ShowSystemWindow (MatterHackers.Agg.UI.SystemWindow systemWindow) [0x00373] in <49bb6fd6740f4762a381356153fe2e02>:0
  at MatterHackers.MatterControl.WinformsSingleWindowProvider.ShowSystemWindow (MatterHackers.Agg.UI.SystemWindow systemWindow) [0x00031] in <16b4831ddba24ef8b2ca5ca9a43af3cd>:0
  at MatterHackers.Agg.UI.SystemWindow.ShowAsSystemWindow () [0x0002b] in <a5c2db3666d84398b1090e2f2dfee474>:0
  at MatterHackers.MatterControl.Program.Main (System.String[] args) [0x0037f] in <e58f04b6857344df9781155f8ee2c0c8>:0

theofficialgman, 2022-02-13:
running into the same issue on ubuntu 18.04 and debian bullseye on arm64 and armhf linux using the mono 6.12.X stable repo as suggested in the help file. Was anyone here able to find a workaround or the source of the issue?

theofficialgman, 2022-02-13:
seems like this workaround worked for me:
https://github.com/MatterHackers/MatterControl/issues/4790#issuecomment-782700811

ashleycawley, 2022-02-14:
I did get this working in the end, I can't remember exactly how, but I remember being massively disappointed to discover you then cannot export the models as a STL from the Linux version, it just simply does not work. So for me ultimately the work-around was moving to alternative modelling software (FreeCAD) to achieve what MatterControl used to do for me. Sad to say the poor Linux support has lost a MatterControl user of me. I reached out to their support and via GitHub but its clear that development on Linux is not a priority for them.

theofficialgman, 2022-02-15:
> I did get this working in the end, I can't remember exactly how, but I remember being massively disappointed to discover you then cannot export the models as a STL from the Linux version, it just simply does not work.

@ashleycawley I guess I have good news for you then. I've exported STLs from the linux version (latest) and it works just fine.

adob, 2023-09-22:
The following fixed it for me:

```
sudo apt-get install mono-vbnc mono-complete
```