MatterHackers/MatterControl #5106: Feature Request

fugazigames, 2021-08-05:
Really great software, and I've been using it everyday for the last couple of months.  Here are a few requests:

- Orthographic view.  This is the single biggest feature that's missing from the software, if nothing else gets implemented but this it will be a huge win.
-  Measurement tool.  I know there is a basic tool that was added, but I never use it.  Moving the points in 3D makes it very difficult to tell if the point is on a level plane.  Adding this to orthographic view also makes dimensioning a massive improvement.
-  Subtracting without removing the part.  If I want to subtract part A from part B there are times I don't want to remove part A, I just want it to "carve out" the area keeping the original part.
- Bevel and Round edges.  This process is way too tedious.  This should be an option as it is with the triangle primitive shape.  Adding bevel or round edges to the box primitive would save an enormous of time. 
- Removing the dimensions from the scale menu.  This is almost a good idea, but in practice I'd rather have the objects properties always on the left side (height, width, depth, color, etc) and if I want to scale that's a menu item.  This was recently added, but should be removed.  Keep the object dimensions within reach, leave scaling as a separate choice.
- Alignment.  This one is more of an observation.  When selecting items to align, the first object I click on should align to the second, right now it's opposite.  I know you can change the alignment parent from the drop down menu, but that's an extra step.  Say this sentence in your head when aligning two objects:  "I want to align this object TO this object."  
- Cancelling tasks.  If I subtract, or combine sometimes it just takes a very long time.  The ability to cancel tasks would be helpful.
- Rotation tool.  Many times while rotating an object the degree text is off the screen.  Centering it the test while rotating an object is very much needed.

I think that about covers my comments.  Thank you again for a great tool... almost there.  :)

larsbrubaker, 2021-08-09:
Thanks for the feedback. Many of these things are in our feature list, and we will add the ones that aren't..

yuval2508, 2022-03-12:
Can you add transparency to objects? This would be great to see the internals while designing