MatterHackers/MatterControl #5123: matter control library continually closing

mereandjim, 2021-09-09:
We have been having this issue for over a year!  We have been trying to get it resolved since about June.  We have tried the suggestions provided to no avail.

So when we open the Cloud Library we get to see our library for about 15 seconds and then it closes.  Whether we are scrolling or whether we just open it and sit there....it closes and goes black after 15 seconds.  This is a real pain!  We have files that we need to go thru in there and cannot access them and get it cleaned up.  

Any suggestions?  Any help....someone...

Meredyth

mereandjim, 2021-11-01:
> If you can send me the user and pw for this account I will try to
> reproduce and fix the issue.
> You can send it to my MatterHackers email:
>
> If that is not an option, can you create the error and send me the time of
> it occurring so that I can look and see if it was captured by our crash
> reporter.

larsbrubaker, 2021-12-06:
Thanks for sending me your login info. I believe I isolated the problem and fixed it.

If you can please download [this](https://mattercontrol.appspot.com/downloads/development/ag9zfm1hdHRlcmNvbnRyb2xyOwsSB1Byb2plY3QY6gcMCxINUHVibGljUmVsZWFzZRiAgPD6j9_zCwwLEgZVcGxvYWQYgIDw2pO3vAgM) build and close this issue if it is fixed.

mereandjim, 2023-02-14:
we are still having issues with Mattercontrol Cloud library

mereandjim, 2024-01-24:
This issue has gotten worse, now i can not get to any of my files.