MatterHackers/MatterControl #5158: The type initializer for 'MatterHacker.MatterControl.SlicerConfiguration.ProfileManager' threw an exception.

JPCMediaLLC, 2021-11-11:
Bug Report
============
**Steps to Reproduce**
1. New Install on MacBook Air
2. 
3. 

**Expected Behavior**
The Program would run.
**Actual Behavior**
Error message at startup 
The type initializer for 'MatterHacker.MatterControl.SlicerConfiguration.ProfileManager' threw an exception.
**MatterControl Build Number**
(To find this, go to the MatterControl menu in the top left and click "About MatterControl")
2.21.10.11188
**Operating System Version**
Mac OS 11.6.1
**Printer Make/Model**
Airwolf Axiom Dual direct drive

Note:  It does run on my Mac Mini - same OS but this computer did have a previous version installed.

larsbrubaker, 2021-11-23:
Making sure you delete any existing versions of MC before installing a new one can be a great help. Sometime Macs keep track of installed applications even when they are not in the applications folder and will run an old version even after an install of a new one.

If that does not fix the issue sometimes deleting the old MatterControl application data can help. [Here](https://help.matterhackers.com/article/206-clearing-appdata) is a guide on how to do that.

JPCMediaLLC, 2021-11-26:
Using Terminal I located and deleted the files.  This fixed the issue.  Thanks.

closed 2021-11-26 by JPCMediaLLC

larsbrubaker, 2021-11-26:
Glad it worked. Thanks for letting me know.