MatterHackers/MatterControl #5175: Pause/Resume Feature Failure after recent update

bryce-laycock, 2021-12-09:
Bug Report
============
**Steps to Reproduce**
1. Inserted a specific layer for my print to pause in the already incorporated "pause /resume" function, so I could perform a material change mid-print.
2. Print paused as it should. I selected "OK" on the window telling me the print paused, then moved my hot end up (Z+) 10mm to perform the material change and purge, then returned it back to its paused position.
3. Option to "Resume" print was no where to be found. the status bar at the bottom of Matter Control still said "print paused," but there was no way for me to tell the software to resume my print. I searched everywhere. I did this exact same operation yesterday before I updated to the most recent software version with no hiccups... Casually waste ~.5kg of material in what shouldn't have been a failed print.

**Expected Behavior**
I expected my print to resume after using the pause on X layer option.

**Actual Behavior**
Print paused as it should, but there was no option to "resume" once I was ready to.

**MatterControl Build Number**
(To find this, go to the MatterControl menu in the top left and click "About MatterControl")
I'm at work and away from my control laptop... I'll update with this info if I can remember.

**Operating System Version**
Most recent Matter Control version operating on the most recent Windows 10 OS.

**Printer Make/Model**
Robo 3D R1+ (that is modified and actually reliable, unlike its stock brethren).

larsbrubaker, 2021-12-09:
Sorry you had a problem with the new build. I tried to reproduce this on the latest build with an R1, but I could not create the error. If you can explain what was different for you we will be happy to investigate and fix the issue.

Here are screen shots of what I was getting:

[image]

[image]

[image]

[image]

larsbrubaker, 2021-12-09:
I did do a move up, filament swap and move down during this procedure.

bryce-laycock, 2021-12-10:
Hi Lars,

Thanks for the reply. On your second image where a window says your print
has paused, then gives you the "resume" or "OK" options, I clicked "OK" and
that window went away. I then did my filament swap using the manual axis
controls. After that was complete and I had moved my extruder back to it's
paused position, that's when I had no option to resume the print again.

Another note: once I clicked out of that initial window, The print status
square (you show in the last image) was also no longer there. The only
remaining indication of a paused print was the lower left corner of the
main UI stating such.

A final note: During my restarted print after I had this issue, my printer
itself started having a hardware error. Perhaps it started failing during
this first print, but not long enough to actually throw the "hardware
error" alert yet? Not sure.

I haven't had a chance to try the next print yet and see if the issue
persists, thanks to me trying to change out a super cheap nozzle and
rolling it in half, leaving me a heater block with a stuck chunk of cheap
brass inside. Once I'm back up and running I'll be sure to respond back if
I can repeat this error, and actually include the build ID.

Cheers,

Bryce

larsbrubaker, 2022-03-03:
Thanks for the update. Looking forward to hearing if this is still a problem.

I will try and reproduce the steps you described.