MatterHackers/MatterControl #5177: Implement Dual Depth Peeling

larsbrubaker, 2021-12-17:
Investigate and implement better transparency for the mesh renderer.

NVidia [paper](http://developer.download.nvidia.com/SDK/10/opengl/screenshots/samples/dual_depth_peeling.html) on depth peeling.

[Here](https://raw.githack.com/pailhead/three.js/depth-peel-stencil/examples/webgl_materials_depthpeel.html) is a great example in WebGL, this is where I would start with understanding a clean implementation.

It is important that this only render the actual 3D scene multiple times and not the rest of the UI. All passes must be constrained to the 3D view.

**NOTE:** Make sure the GLFW renderer is active when solving this issue.  **Enable Line 234 in Program.cs**

fortsnek9348, 2023-07-18:
These bounty labels disappearing doesn't seem great. Is there still a bounty on this one?

Some of this stuff might need multiple passes, so there may need to be a higher level rendering interface passed to all these drawables.

Also would like to fix some things up. Deduplicate some code maybe.

Vulkan is a fun idea, but it's a massive change in how things are done. But well, if the raw graphics API calls are moved out of application code, it would be more feasible to flip flop between graphics APIs.

larsbrubaker, 2023-07-19:
Sorry about that. We got rid of our bounty program as there was very little participation and the overhead of it has not been worth the time. We would still love the contribution but it would have to be for the good of the project. :)

Thanks for all the work you have done and all the interest you and everyone has had in the project.